portapack-mayhem/mayhem-firmware #1889: BLESpam App Causes Crashes

tomzhu1024, 2024-02-13:
**Describe the bug**
When opening the BLESpam app, my device crashes with M0 Guru Meditation.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Transmit'
2. Go to 'BLESpam'

**Expected behavior**
It should open the app.

**Affected versions**
* Latest Nightly release: n_240212

NotherNgineer, 2024-02-13:
Thank you for reporting this.  It is a known issue, duplicate of #1879, and it should be resolved in tonight's nightly build.  I'm going to mark this one as a duplicate, but please try the n_240214 build after it comes out (in about 4 hours) and let us know if there's still an issue.

closed 2024-02-13 by NotherNgineer

tomzhu1024, 2024-02-14:
Thanks for the answer! I just tried the `n_240214` build, it works nicely now.

NotherNgineer, 2024-02-14:
@tomzhu1024  Thank you for confirming that the problem is fixed.