portapack-mayhem/mayhem-firmware #1901: I want know how create process,to smaple capture

girs1982, 2024-02-15:
I successfully connected and accepted some sample using
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Access-Radio-Hardware
this sample, (proc_afskrx.cpp)
but I want to create my own process, and understand how it
it works
/* NB: Threads should be the last members in the class definition. */
BasebandThread baseband_thread{baseband_sampling_rate, this, baseband::Direction::Receive};
i found
constexpr image_tag_t image_tag_acars{'P', 'A', 'C', 'A'};
constexpr image_tag_t image_tag_adsb_rx{'P', 'A', 'D', 'R'};

constexpr image_tag_t image_tag_nrf_rx{'P', 'N', 'R', 'R'};
spi_image.hpp in this file,
you need a more detailed wiki, because I do not know how to create my own process, and what there is responsible for, it would not be superfluous to describe the interaction normally!
a simple example of how to take a sample buffer in your process
[image]

girs1982, 2024-02-15:
i almost understood how it works, but in order not to spoil the process proc_afskrx.cpp&hpp,ya I want to clone it under my own name, and then remake it for myself, please write how to do your process for example image_tag_my_rx,
so that I season in the falls and it is a duplicate of afskrx, I don't seem to ask much

htotoo, 2024-02-18:
Check this commit: https://github.com/portapack-mayhem/mayhem-firmware/commit/02251eeeb4fce89e5e3bc50113030c50bcb4a05f

In it, I added a new one. Forget everything in the fproto folder, and just check the other files, and it won't seem so much.

girs1982, 2024-02-18:
ok , but it very hard code style, need some simple ,how decode ook codes

girs1982, 2024-02-18:
> Check this commit: [02251ee](https://github.com/portapack-mayhem/mayhem-firmware/commit/02251eeeb4fce89e5e3bc50113030c50bcb4a05f)
> 
> In it, I added a new one. Forget everything in the fproto folder, and just check the other files, and it won't seem so much.

ok , but it very hard code style, need some simple ,how decode ook codes

sizzlik, 2024-02-18:
Maybe take a look at URH here on github for decoding ook. Great tool.

girs1982, 2024-02-18:
> Maybe take a look at URH here on github for decoding ook. Great tool.

I have not tried such a tool yet, I will definitely try.

girs1982, 2024-02-18:
> Check this commit: [02251ee](https://github.com/portapack-mayhem/mayhem-firmware/commit/02251eeeb4fce89e5e3bc50113030c50bcb4a05f)
> 
> In it, I added a new one. Forget everything in the fproto folder, and just check the other files, and it won't seem so much.

it all very many code, need simple get samples and modulations setup, it all

girs1982, 2024-02-18:
came work in suGz but starline and keeloq not work

girs1982, 2024-02-19:
data bits not rounded and serial as FFFF , i think i can fix it

girs1982, 2024-02-19:
all memory full on device ,what the animal-fuckers shit, need remove teris and pacman

zxkmm, 2024-02-19:
Games are not loaded if you don’t launch them. So they don’t eat any of the memory.  
Please note you language and being respectful to developers of games, otherwise i will close and lock this issue ticket.
Thank you.

closed 2024-02-19 by zxkmm