portapack-mayhem/mayhem-firmware #1912: Update error

samukas81, 2024-02-17:
Hello, it is not updating via the WEB from the stable version 2.0 to the last night.

gullradriel, 2024-02-17:
Because there has been no nightly since the 2.0 ?

samukas81, 2024-02-17:
17/02/2024?

jLynx, 2024-02-17:
What version are you currently on and what version are you trying to update to on MayhemHub? Also side note, this issue should possibly be opened on https://github.com/portapack-mayhem/MayhemHub/issues instead of this repo

samukas81, 2024-02-18:
2.0 to 17/02/2024

jLynx, 2024-02-18:
Thanks @samukas81, what error are you seeing and what's happening with the device when you try to do it?

samukas81, 2024-02-18:
It doesn't show any error, it downloads, uploads to portapack but doesn't update, doesn't change the version.

jLynx, 2024-02-18:
Do you see the update screen at all on the PortaPack? Also are you on a mac or pc?

samukas81, 2024-02-18:
Yes, I see every process normally, I use w11 64bits

jLynx, 2024-02-18:
Are you on our discord? If not, could you please join and message me there, will be faster to get this bug sorted out for you

jLynx, 2024-02-18:
https://discord.gg/XVqEvESTem

samukas81, 2024-02-18:
Don't worry, I just reported the error so that you can follow it and find a solution, thank you very much for your feedback and help!

jLynx, 2024-02-18:
Would you mind creating the issue on the other repo I mentioned above so then we can close it here

samukas81, 2024-02-18:
Okay, I'll create

samukas81, 2024-02-18:
I can't access my account, I'm sorry.

jLynx, 2024-02-18:
Its the same account you use to reply here

closed 2024-02-18 by jLynx