portapack-mayhem/mayhem-firmware #1914: M0 GURU MEDITATION ERROR

lucidliving69, 2024-02-17:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
When opening RECEIVE -> RECON -> CONFIG -> SELECT INPUT FILE I get a M0 GURU MEDITATION error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'RECEIVE'
2. Tap on 'RECON'
3. Tap on "CONFIG"
4. Tap on "select input file"
5. CRASH with M0 GURU MEDITATION "HINT: Out Of Memory"
6. "Press DFU for Stack Dump" (included in video)

**Expected behavior**
Able to select input file from SDCARD.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 2.0.0
* Latest Nightly release:
* Previous working release: 1.9.3

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.
https://github.com/portapack-mayhem/mayhem-firmware/assets/135712432/9cd64610-e2db-4603-b56b-c42545801daa

[image]

gullradriel, 2024-02-17:
It may be that you have too much files on the directory you are trying to open.

lucidliving69, 2024-02-17:
what's the max amount of files allowed per folder?

The SD card is a 256gb SanDisk

gullradriel, 2024-02-17:
I could reproduce.
It's not triggered each time.
It may be due to last freqman modifications, or optimization things.
Poke @NotherNgineer at least.
We should rewrite that file manager app so it's not eating the memory.
IIRC it's maintaining a list of current directory files instead of a buffered view of the entries.

NotherNgineer, 2024-02-18:
Same issue as in #1752 and #915 but the number of files before this happens (and available RAM) keeps decreasing as new firmware features are added.  I recommend keeping this one open until we improve File Manager to handle more files, or reduce the app size.

lucidliving69, 2024-02-18:
Sounds good, I'll keep it open until one of you tell me it's solved.

NotherNgineer, 2024-02-18:
Some random heap space measurements, starting with 32KB M0 heap space available when viewing the Main Menu:

*  Opening the Utility Menu consumes about 4KB (about 300 bytes per app button), opening File Manager from the Utility Menu consumes another ~6KB of heap space, and then opening a directory consumes around 182 bytes per file.

* Opening the Receiver Menu consumes about 5KB, opening Recon from the Receiver Menu consumes another ~9KB (depending on operating mode), opening the Config Menu consumes another ~3KB, opening "Select Input File" (File Manager) consumes about ~12KB with the 57 default FreqMan files.  (leaving less than 3KB free)  Actually loading the frequency file consumes additional memory.

If File Manager were modified to only show the number of files that fit on the screen (and only fetch the rest when scrolling or changing directories), it would probably save about 8-9KB of heap space, preventing this issue.

gullradriel, 2024-02-18:
I'm totally for the file manager optimisation.
Maybe I'll have time to do it, maybe not.

gullradriel, 2024-02-18:
Nice investigations BTW

NotherNgineer, 2024-02-19:
Another way to save some memory might be to deallocate the menu buttons when opening an app (of course the menu screen would then need to be reconstructed when the app exits but we'd just need to remember which app the focus was on).  That would save about 5KB in the case of the Receive menu, providing more working space for all receive apps, but only freeing up enough space for about 25 more files in File Manager.  Probably we should implement both of these suggestions.

BuddyPage, 2024-02-20:
Is that a new SD card? Did the problems start with it? Try a 32gb and see if the errors stop. I had to downsize from 64 to 32 in 1.9.1 to stop freezing and other errors, and it still works with no similar problems in 2.0. I had a bunch of problems that stopped when I solved my read errors by downsizing to a 32gb SanDisk. Does the 256 work for you now?

lucidliving69, 2024-02-21:
no, I've had the same sdcard since I got the device last year

the card works perfectly fine, adding folders to FREQMAN fixed the issue

BuddyPage, 2024-02-21:
Thanks. I'll try a larger SD card again soon and see if my problems return. In fact, I'll go with a new 256gb Sandisk.

lucidliving69, 2024-02-21:
256gb Ultra A1

closed 2024-03-10 by gullradriel