portapack-mayhem/mayhem-firmware #1920: M0 Guru Meditation - Out of Memory - Soundboard.

StefCoders, 2024-02-19:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

I keep getting a M0 Guru Meditation - Out of Memory on the soundboard.

**To Reproduce**
Steps to reproduce the behavior:
1. Newest nightly firmware
2. Tap on transmit and soundboard (i have custom files.)
3. Error occurs.
4. 

**Expected behavior**
A clear and concise description of what you expected to happen.
Open the soundboard app, and be able to play files.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
Latest Stable release:
Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2024-02-19:
How many files do you have in the WAV folder?  Sounds similar to #1914.

StefCoders, 2024-02-19:
Sorry for the late reply but I have around 91 items converted with FFMPEG. Works on v1.9.1

NotherNgineer, 2024-02-19:
Firmware version 2.0.0 uses more memory so you'll probably have to limit the file count to a fewer number than in 1.9.1 until a real fix is implemented.  Please let us know if that works, and how many files you had to remove.

BTW, removing some of the Main Menu or Transmit apps from the menu can also help (you can probably add about 2 more WAV files for every Main Menu or Transmit app that's hidden, either by creating a "blacklist" file, or by deleting the .ppma file from the APPS folder in the case of external apps)

StefCoders, 2024-02-19:
Ahh thank you! 

Is there any like limit? or is it just my specific situation.

I'll close it and thanks!

closed 2024-02-19 by StefCoders