portapack-mayhem/mayhem-firmware #1926: Soundboard stuck & GURUs when 80 files in WAV folder

unknown10777, 2024-02-22:
**Describe the bug**
Soundboard stuck & GURUs when 80 files in WAV folder

**To Reproduce**
Steps to reproduce the behavior:
1. download bunch of music & transcode to wav
2. download them to SD card's WAV folder
3. boot portapack, click on soundbrd
4. GURUs

**Expected behavior**
I expect it won't crash like WAV viewer's

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 240114..240211

eried, 2024-02-23:
does not happen if there is less than 80 files?

unknown10777, 2024-02-25:
65 is OK, but 70 crashed

unknown10777, 2024-02-25:
I think there may be unclean GC in file manager and soundboard

NotherNgineer, 2024-03-21:
FYI, PR #1971 should have freed up a little memory to allow a slightly larger file count, but File Manager still needs work to only require memory for the files shown on the screen versus for every file in the folder.

closed 2024-03-23 by gullradriel