portapack-mayhem/mayhem-firmware #1936: Inquiry about Automatically Opening "Search" App on PortaPack Mayhem Startup

Petros04, 2024-02-25:
Dear PortaPack Mayhem Authors,

I am currently exploring the features of PortaPack Mayhem and am interested in customizing its startup behavior. Specifically, I would like to know if there is a way to automatically open the "Search" app whenever the PortaPack is turned on.

I believe this feature would greatly enhance the user experience by providing quick access to the search functionality right from startup. Could you please advise on how this customization can be achieved?

Thank you for your time and assistance.

jLynx, 2024-02-25:
I think this is a cool idea. It's not currently possible, but shouldent be too hard to do

Petros04, 2024-02-26:
> I think this is a cool idea. It's not currently possible, but shouldent be too hard to do

Can you help me to do it now ?

Sliicy, 2024-03-13:
#1981 similar idea but to have Pacman open automatically so the PortaPack looks innocuous.

PR #2079 (2024-04-03): Autostart option
In the settings there is a new tile, where you can select what apps should show on startup.
Resolves #1936
Resolves #1981
closed 2024-04-03 by gullradriel