portapack-mayhem/mayhem-firmware #1938: create my own Word map Help

Hackrfan, 2024-02-26:
Hello to the team,
I want to create my own Word map. Unfortunately I have problems with Python 3. I work with WIN 10 prof but have no idea how to run the Python script under Windows 10.
I have followed the description "https://github.com/portapack-mayhem/mayhem-firmware/wiki/World-map-generation" and created a map.
But I don't know how to execute the py script. I would appreciate some help.
George

amelillo, 2024-03-05:
Me too.
I mean, I'd like to generate a detailed world map but just for the part of the world I live in (Italy) so that I could have some good resolution when I zoom in.
Excluding the rest should help to keep the size small so that the HackRF doesn't hang.
No problems with python here (linux machine).
Exactly, what problem do you have with the script?
You're supposed to run the script in a terminal (cmd window) cd'ing to the dir where the map and the script are stored.

Hackrfan, 2024-03-09:
hello,
thanks for the info. I have now managed to create the bin from a map. My mistake was that I had not yet included the "Pillow" in Python and then I adapted the script to my folder and executed it.
Unfortunately the coordinates in the generated World_Map are not correct. I do not yet know how to correct this.
Maybe someone has an idea why. The representation in the Portapack does not match the real data.

Translated with DeepL.com (free version)

NotherNgineer, 2024-03-19:
I have noted one more rule in the world map instructions on the wiki:

-   The center of your map must be 0-degrees latitude and 0-degrees longitude.

Does that help?

Hackrfan, 2024-03-21:
Where is that written ?

0 /0 is too far from my map center.

NotherNgineer, 2024-03-21:
I added this info to the same wiki page you mentioned at the top of this issue.

That's kinda why it's called a "world" map.  Code is currently assuming it's the whole planet and not just one area.  I'd have to think about how to modify the geomap code to handle a smaller area, and how you might be able to tell the firmware about your partial map.

amelillo, 2024-03-22:
That would make much sense. None of us is so lucky to be able and travel all around the world to tinker with his HackRF; instead, everyone spends his time in a small portion of the planet.
This also relates to my request to be able to geotag recordings.
Using a smaller map could be useful because geotagged recordings would be displayed at a different zoom level (i.e. street level, for instance) turning the map in a pretty useful tool. Imagine walking and wardriving for hotspots, gates, remotes, whatever.
I mean, displaying a signal in a map which covers three or four hundreds km per side is actually no more than a clever exercise, as a map is useful to locate a place (a signal, for a receiver) and not to show a landscape.
What's perfect to trace a plane or a ship is not to locate a signal.
Btw, almost none has a GPS module, so I fear it's not worth the effort.
However, maybe a smaller map would free up memory and make everything more responsive.

Hackrfan, 2024-03-22:
I think this is interesting for many types of modems to display a smaller map section, e.g. on APRS or weather probes which also transmit the coordinates.
To better display the signals locally.

NotherNgineer, 2024-03-26:
fyi, @htotoo has documented this geomap improvement suggestion in enhancement request #2044

htotoo, 2024-03-26:
Whops, didn't followed this issue, so my may be a dupe of this.

gullradriel, 2024-04-06:
Looks solved to me.

closed 2024-04-06 by gullradriel