portapack-mayhem/mayhem-firmware #1947: Decode digital signals like DSDplus

T1000-Cyberd, 2024-03-05:
### Description of the feature you're suggesting.

Decode digital signals like DSDplus

phish27134, 2024-07-05:
You'd have to write a ton of code, I don't know the native language of the rf but bc its older id say C or Java......dSD has a ton off dlls

gullradriel, 2025-02-04:
Decoding DRD (Differential Radio Data) and DSP+ (Digital Signal Processing Plus) directly on the HackRF Portapack is likely to be very challenging due to hardware limitations. Here's why:

# 1. Hardware Constraints of HackRF Portapack
- Limited Processing Power: The HackRF One (even with the Portapack Mayhem firmware) relies on an ARM Cortex-M4 microcontroller (STM32F4). This MCU is great for handling control logic but lacks the raw DSP power needed for real-time signal decoding beyond simple modulations.
- No FPGA or DSP Accelerator: Unlike SDRs like the USRP or LimeSDR, the HackRF doesn't have an FPGA or dedicated DSP processor. It relies purely on software processing.
- Memory Limitations: Your firmware already faces space constraints (4668 bytes left in flash). Complex demodulation might require additional libraries or look-up tables that may not fit.

# 2. Feasibility of DRD and DSP+ Decoding
## DRD: If it is similar to other differential encoding schemes, decoding could require:

- Synchronization to a reference signal.
- Phase-difference calculation over time.
- Potential error correction (which might need extra processing).

If DRD is purely differential and relatively simple, it might be possible with heavy optimization.

## DSP+: If this refers to an enhanced DSP modulation/demodulation scheme, it might involve:

- High-order QAM or PSK demodulation.
- Advanced equalization and filtering.
- High-speed FFT or convolution operations.

These typically require floating-point operations, which the STM32F4 lacks in hardware (though it supports software floating-point).

# 3. Potential Workarounds
- Preprocessing on a Host PC: Stream raw IQ data from the HackRF to a PC and handle the decoding there.
- External Co-processor: If the decoding is essential in an embedded setup, an ESP32-S3 or an FPGA could assist with DSP tasks.
- Highly Optimized C/C++ Implementation: Stripping down the DSP processing to integer math and LUT-based calculations might make a basic implementation feasible.

# Conclusion
Unless DRD and DSP+ are very lightweight modulation schemes, real-time decoding directly on the HackRF Portapack is unlikely to work efficiently. Offloading the heavy lifting to a more capable processor is probably the best approach.

closed 2025-02-04 by gullradriel