portapack-mayhem/mayhem-firmware #1948: Radiosonde M20 additional params

voxo22, 2024-03-06:
### Description of the feature you're suggesting.

Hi, is it possible to add temp, humidity, pressure, batt, etc params to existing ones for Meteomodem M20?
Thank you!

joyel24, 2024-03-09:
Hello @voxo22,

I added the feature to identify & decode M20 coordinate and altitude because this was the only things needed to chase them and didn't really care about temp/hum/pressure/Vbat...

If you wish to give me a bounty to add this feature I can to it for you.

voxo22, 2024-03-09:
How much would you like for that and also for adding detection for the DFM probe? I'll see accordingly

joyel24, 2024-03-10:
Unfortunately I won't be able to work on decoding DFM as there isn't such a sonde in my country.

For M20, It depends on you how much you want to offer for this feature ?

voxo22, 2024-03-10:
Hm, I have no idea how much effort/time can this take. 20€? I have DFM 17 sonde here, I can share IQ file with you

joyel24, 2024-03-12:
I don't think I have enough skills to reverse it only with an IQ file.
Would be easier if I had a 2nd hackrf but it's not the case...

I can start to work on decoding other data from M20 if the ammount you wrote is ok for you without RS41 ?

voxo22, 2024-03-12:
ok, I understand. Yes, the amount for M20 decoding is ok for me, if it's ok for you....

voxo22, 2024-03-12:
I think you should definitely consider buying 2nd hackrf to comfortly simulating any recorded signal ;-)

voxo22, 2024-04-26:
@joyel24 hi, any update? Do you think it is doable?

voxo22, 2024-05-26:
@gullradriel @joyel24 I have successfuly made LOG decoder (TEMP, HUMIDITY, PRESSURE, BAT_VOLTAGE) for this M20 sonde, so now I can support direct sonde_packet.cpp code update. I can post it here, can you please move it then to original code and compile? I am too lazy to do that :-/

joyel24, 2024-06-03:
> @gullradriel @joyel24 I have successfuly made LOG decoder (TEMP, HUMIDITY, PRESSURE, BAT_VOLTAGE) for this M20 sonde, so now I can support direct sonde_packet.cpp code update. I can post it here, can you please move it then to original code and compile? I am too lazy to do that :-/

Sorry, at lot busy... I don't know if you still need it, I did not understood well your last comment.
What do you mean by you made log decoder ? If you've been able to do it, why don't you open PR ?

voxo22, 2024-06-03:
@joyel24 I have made it as webpage in PHP. M20 sonde has very easy-to-decode bitstream, so I think adding missing params to sonde_packet.cpp  is quite easy. But I am not familiar with C++ so much plus I have no compilers installed, so if you provide me with the way how to modify/add the code in this repo, I can do it. Who will test it then (and make final firmware release)?

Or, you can read my code for M20 (attached) and copy the parts, which are about decoding bat, temp, humidity and gps time.
[M20-RS41.zip](https://github.com/user-attachments/files/15529666/M20-RS41.zip)

voxo22, 2024-06-03:
@joyel24 and one more thing would be nice: Auto-sync of the portapack-clock based on real GPS time from the radiosonde. I have this parameter decoded for both M20 and RS41 (see variable $gpx in my code).

gullradriel, 2024-06-03:
Nice, we have to make something with this

joyel24, 2024-06-03:
Last light I started to work with temperature. I had troubles using my docker building image but I've been able to get the temperature with a caltulator using the formula in the radiosonde_auto_rx project.
I've built this firmware that normally decodes temperature, but I'm not can't check because I have nothing actually that can decode temp of the M20 to check if it's matching.

The bin is here:
https://github.com/joyel24/portapack-mayhem/releases/download/temp/portapack-h1_h2-mayhem.bin

voxo22, 2024-06-03:
@joyel24 I have tried your bin. There is temp displayed, but, in comparison with the log, there are kind of weird values. Normally, now is 20.3°C, radiosonde app showed 19.6. When I touched the sensor, there is pretty curve up to 33°C and down in the log, app shows still 19.6. There was one more value, 22.3 in the app, but it seems like only these two values are possible, nothing else.

joyel24, 2024-06-03:
Yep look like temp were good but with big steps. Added .0 to numbers and it looks better now : https://github.com/joyel24/portapack-mayhem/releases/download/temp/portapack-h1_h2-mayhem.bin

voxo22, 2024-06-04:
Yep, granularity of the value is now ok. But you definitely compute temp from the wrong parameter. You use get_Tntc2 function (see lines from 636 in attached zip), instead of get_Temp function (see lines from 564). Thats why temp sensor does not respond to my hand touch. I mean get_Tntc2 is kind of temp measurement, which use humidity sensor or some other magic to acquire data (maybe it shows dew point temp, or some kind), but definitely not temp sensor.
[c_m20mod.zip](https://github.com/user-attachments/files/15543045/c_m20mod.zip)

voxo22, 2024-06-21:
@joyel24 Hi, do you think it is possible to finalize the code in short time? M20 also supports air pressure [hPa] data, so all three of them (temp/rH/pressure) could be displayed simultaneously. if needed, please use excerpt from the c_m20mod.zip (c++) above.

joyel24, 2024-12-07:
Hello, I did not work on RS this year finally...
I was working on other stuff.
I thought someone did this fix but I just realized it's still open. Maybe I can try before spring 2025 if not done yet...
Do you still need this fix ?

voxo22, 2024-12-07:
@joyel24  of course I would welcome the fix. The Radiosonde app has several more errors, e.g. a truncated timestamp, missing pressure and dew point data for the RS41 sonde, but the M20 displays virtually nothing, except for the "Meteomodem" and the position that you added. The entire program in C++ is available and you just need to rewrite the procedures into the code in the mayhem app, just like I did in PHP when I was making a decoder for the Radiosonde log. There are already a lot of M20s flying in the world today, so it's a big shame that the data can't be monitored in real time on the display, but only from the log. The M20 provides virtually the same data as the RS41, so there could also be displayed ID of sonde, battery voltage, temperature, dew point, humidity, and pressure. They are all available in C++ code, which I also uploaded to the issue. If I could compile it myself, I would have done it, but I'm completely out of my depth when it comes to compiling for mayhem.

htotoo, 2024-12-08:
Anyone from here could test this fix for m10: https://github.com/portapack-mayhem/mayhem-firmware/issues/2342
?

htotoo, 2024-12-27:
Can anyone provide a log generated with portapack, and maybe a c8 (or c16) recording?
That could help with the project.

joyel24, 2025-01-20:
> Can anyone provide a log generated with portapack, and maybe a c8 (or c16) recording? That could help with the project.

It's mostly french but I did this reverse engine to the frame, it contains mistakes
https://github.com/joyel24/m20-playground/blob/master/Trames%20M20%20hexa.txt

voxo22, 2025-01-20:
For sure, here they are (both for M20 and RS41). I have added my PHP code to parse logs. Also C8 for DFM17.

[sonde.zip](https://github.com/user-attachments/files/18482741/sonde.zip)

Summarized what should be done @portapack:
1. fix timestamp display (chopped off), better would be showing real date & time for better readability
2. fix showing temperature decimals sequence (now displays e.g. ...0.8, 0.9, 0.10, 1.0, 1.1.... ), so 0.10 will not be in it
3. add remaining additional parameters of RS41 from the log on the screen (would be nice to have Pressure & Dew Point)
4. complete/review/rewrite M20 screen (missing ID, temp, batt, humidity + additional parameters, based from RS41, except Calibration blocks - M20 does not have them)

voxo22, 2025-11-23:
@htotoo Hi, would you like to take a look at this modification as well? Just copy the decoding code of the M20 parameters from the code I uploaded. Thanks for all the modifications in the Radiosonde app so far!

htotoo, 2025-11-23:
Sorry, but it is not just a simple "copy", it has different structures.
Also i don't have any m20 sondes, just the rs one, so also i can't really test. 

but if you could send screenshot froe example for the problem # 1, or describe (or show screenshot of the current, and the expected) of # 2.

for the 3: i will look into it, and maybe show it in the details screen, but iirc the pressure is now working?!

htotoo, 2025-11-23:
For the easier fixes, the screenshot and the explanation would be helpful. 
for the other parts we should try with a good c8 capture.