portapack-mayhem/mayhem-firmware #1966: Adding geotag feature to record app

amelillo, 2024-03-10:
### Description of the feature you're suggesting.

I was thinking about the recording app and the GPS capabilities of the HackRF.
Wouldn't it be useful to be able to geotag a recording, right after having saved it to the sd card, to be able to reference it withouth the need to choose a proper filename?
Then, we could even export the recordings and the geotag data to display recorded signals on a map.
Especially useful when dumping subGHz signals of gates, remotes, doorbells and similar.

htotoo, 2024-03-13:
It is doable, will look into it.
For now I think, adding it to the txt (lat=23.2422 lon=2.3444) format should be enough, and maybe a show on map if the geo info present.

amelillo, 2024-03-13:
Sure. The single recording should be referenced in the txt file (ex. RECORDING01.C16 LAT=XX.XXXX LON=XX.XXXX), one recording per line.
Showing them on map might be a waste of programming time as it would be really useful only if the map could zoom at street level.

htotoo, 2024-03-14:
Map can't do (by default) the street level zoom, so showing signals in a map wouldn't be effective.
My idea is when i got a recording01.c16, that has a recording01.txt too, and it a gps module is present, then it would store the coordinates too.
Could check the location per recording on the map.

But an external app could load all recording txt-s, and parse them, and show them on a map. So it wouldn't grow the base apps too big, and won't confuse users (who don't have gps, why there is a gps list button).

htotoo, 2024-03-28:
Linked the displayer ext app, so if merged this will be closed.
If want to request more features for this topic, please open a new issue.

closed 2024-03-29 by NotherNgineer