portapack-mayhem/mayhem-firmware #1973: Add audio "beep" support

NotherNgineer, 2024-03-10:
### Description of the feature you're suggesting.

Reminder to myself to add support to baseband for outputting an audible tone of a specified frequency to the speaker/headphones (preferably sine wave).

To be used by apps in the future to represent signal strength (higher tone frequency = higher signal strength).

RomanEmelyanov, 2024-03-13:
What we need to do to make a beep via speaker/audio_jack from UI app?
I'm trying to add "baseband::request_beep();" to LookingGlass app (for make a noise on specific power level) , but it's not enough ...

NotherNgineer, 2024-03-13:
At the moment, only proc_sonde supports baseband::request_beep(), for balloon tracking.

For this enhancement suggestion I was hoping to implement something a bit nicer, but I haven't finished it yet.

For now, one option might be to port the beep code from proc_sonde over to proc_wideband_spectrum (background process used by the Looking Glass app).

RomanEmelyanov, 2024-03-14:
For debug purpose I'm add to proc_sonde.cpp two lines of code:

`    check_beep.on_select = [this](Checkbox&, bool v) {
        beep = v;
        baseband::request_beep(); // make beep then press 'BEEP' checkbox
        nav_.display_modal("Beep", "Did you listen beep?"); // show question
    };
`

but no voice via speaker/audio_jack, any suggestion ?

NotherNgineer, 2024-03-14:
You are correct; looks like the old beep code doesn't work.

NotherNgineer, 2024-03-19:
@RomanEmelyanov, fyi, PR #2012 should fix the Sonde RSSI Beep support, which you might now be able to use as an example for Looking Glass (see the audio::dma code in proc_sonde)

closed 2024-03-19 by gullradriel

RomanEmelyanov, 2024-03-25:
@NotherNgineer beep in sonde app works perfect)

But I have some troubles with beep in Looking glass app.

I'm add to file 'proc_wideband_spectrum.hpp' this peace of code:
```
void WidebandSpectrum::on_message(const Message* const msg) {
...
        case Message::ID::AudioBeep:
            on_beep_message(*reinterpret_cast<const AudioBeepMessage*>(msg));
            break;
...
void WidebandSpectrum::on_beep_message(const AudioBeepMessage& message) {
    audio::dma::beep_start(message.freq, AUDIO_SAMPLE_RATE, message.duration_ms);
}
```
and try to invoke 'baseband::request_audio_beep()' from 'GlassView::add_spectrum_pixel'@ui_looking_glass_app.cpp

But no sound, I'm something missing to add?

gullradriel, 2024-03-25:
Looking glass is already having beep sound. You can check the source in next.

RomanEmelyanov, 2024-03-25:
> Looking glass is already having beep sound. You can check the source in next.

I'm use the old code, Thanks a lot!

PR #2012 (2024-03-19): Fixed Sonde RSSI "Beep"
1.  Added audio output beep support to the baseband audio_dma module.
2.  Cleaned up and fixed RSSI Audio Beep in Sonde app, whereas before it was silent.
3.  Produce a beep sound when enabling Beep (on packet reception) in the Sonde app.
4.  Cleaned up dead code in the background tone_gen module.

I haven't actually verified that the tone frequencies are suitable for various RSSI levels, but at least it's making a beep sound now.  Will check beep tone frequency later.  Note that the volume (in upper right corner) may need to be increased from the default.

Fixes #1973
 [... cut]