portapack-mayhem/mayhem-firmware #1978: "This site cant provide a secure connection"

howdid1gethere, 2024-03-11:
### Describe the bug.

 Using Chrome, navigating to https://hackrf.app yeilds the following error: 

[image]

Am i missing something?

### Reproduction

1. Plug in Portapack via USB
2. Navigate to https://hackrf.app
3. Receive Error. 

Alternatively, 

Navigate to hackrf.app without usb plugged in
Receive Error

### Expected behavior

1. Plug in via usb
2. navigate to https://hackrf.app
3. load into PWA

gullradriel, 2024-03-11:
No problem here, try to clear your cache / stop your VPN ?

gullradriel, 2024-03-11:
I've just updated two devices right now, so it's working.

howdid1gethere, 2024-03-11:
It appears to be something to do with the Internet connection at my current location. Checking with my partner it appears to be working where they are.

closed 2024-03-11 by howdid1gethere