portapack-mayhem/mayhem-firmware #1979: Audio over USB

tony1tf, 2024-03-11:
### Description of the feature you're suggesting.

Now we have USB to serial, is anyone working on audio over USB? This would allow the Portapack to be used remotely for both apps and audio out and in. Is the processor powerful enough to do this? Modern ham radio transceivers all have these abilities, but obviously cost a lot more than the Hack-RF and Portapack.

sa7mon, 2024-03-17:
You could set the PortaPack to HackRF mode, connect it to a laptop/desktop, and run [rtl_tcp](https://manpages.ubuntu.com/manpages/trusty/man1/rtl_tcp.1.html) to start a server. Then connect to it from any device on your network. But at that point, why are you using a PortaPack at all when a plain HackRF will do.

tony1tf, 2024-03-17:
Thanks sa7mon.  The rtl_tcp info shows it as being suitable for use with RTL2832 tuners as used in rtl-sdr. Does HackRF emulate this, including transmit mode? 
The PortaPack gives me the best of both worlds by making HackRF portable while still being able to use HackRF in its original USB tethered mode.
Tony

sa7mon, 2024-03-17:
I’m not sure, but it’s very likely. Only one way to find out!

gullradriel, 2024-04-06:
Usb serial will not be quick enough to stream audio.
Or at least not how it is actually. 
Better directly use the audio jack to stream audio.

closed 2024-04-06 by gullradriel