portapack-mayhem/mayhem-firmware #1980: (Vote) Fine tune of sdcard structure and naming

zxkmm, 2024-03-12:
### Description of the feature you're suggesting.

I listed a rough list below. You can click "Edit" button to edit new options or write pros and cons.

### Anything else?

todo: take care of FIRMWARE and APPS

# way 1 (u-foka)

* rename blob folder to /SYS, and put /FIRMWARE and /APPS into it
* add /FW_BACKUP under /USR

pros: The logic is more clear

cons: User would not be able to easily update/upgrade blob files by simple delete and paste new one

# way 2 (HTotoo)

* let /FIRMWARE stay root
* rename blob folder to /SYS, and put /APPS into it

pros:

cons:

# way 3 (u-foka + spamnoob)

* keep blob folder name /RES
* move /APPS and /FIRMWARE into `root/BIN`

pros: user can easily update/upgrade blob files by simple delete and paste new one

cons: root dir looks complicated

# way 4 (spamnoob)

* keep blob folder name /RES
* keep /APPS and /FIRMWARE stay in root

pros: user can easily update/upgrade blob files by simple delete and paste new one

cons: root dir looks extremely complicated

htotoo, 2024-03-13:
my vote: w2, or w4

NotherNgineer, 2024-03-14:
Personally I like "SYS" better than "RES"

NotherNgineer, 2024-03-15:
Regarding making SD Card content updates easier for the user, please also see my related suggestion #1996.

NotherNgineer, 2024-03-28:
Also see another alternative in #2057 - configurable root folder