portapack-mayhem/mayhem-firmware #1985: ADSB and AIS not working again

Herro95, 2024-03-13:
### Describe the bug.

We already had the problem in firmware version 1.8.0 where there were no entries in ADSB and AIS. Now the problem is still there. In the latest nightly version ADSB works but unfortunately AIS does not either. 

### Reproduction

Open Apps and search for signals

### Expected behavior

doesn't work and no entries show up

zxkmm, 2024-03-13:
Thanks for the report, just FYI for other developers, this is not related to #1967 and #1968 as it haven't been merged when this ticket created.

gullradriel, 2024-03-23:
Check that external TCXO setting is not disabled in settings

Herro95, 2024-03-24:
> Check that external TCXO setting is not disabled in settings

changed that option but its still not working.

gullradriel, 2024-03-25:
Perform a factory reset.
Clear all files in SETTINGS.
Update CPLD.
Flash nightly via web flash.
Reboot.
Check settings (external TXCO not disabled)
Set settings, use pmem to sd if your internal button battery is dead

gullradriel, 2024-04-06:
Tested last nightly on 3 devices. ADSB and AIS ok.
Make a clean install like stated before.

closed 2024-04-06 by gullradriel