portapack-mayhem/mayhem-firmware #1988: Mayhem H1 portapack can't change audio volume

pe2ben, 2024-03-13:
### Describe the bug.

In the Receive/Audio App i can't change the volume and the mode,

### Reproduction

switch on
select "receive" then "audio" then mode or volume => i can't change values

### Expected behavior

i would can change the mode and the volume value

### Environment/versions

version 0.19 and also 2.0

NotherNgineer, 2024-03-13:
Sorry I'm unfamiliar with the old H1 model.  On H1, how do you scroll through the options for any field?

pe2ben, 2024-03-13:
Hello Mark, this is a user error (i play a long time now with the H2 version and forgot the correct usage from the H1. Case can be closed. Thanks Ben