portapack-mayhem/mayhem-firmware #1997: Audio RX performance and settings display corrupted

katherinehackworth, 2024-03-16:
### Describe the bug.

DFU button-activated performance and radio settings displays are garbled in audio RX app. Both display incorrectly with the text flickering and following the waterfall off screen, and audio output sounds robotic and low nitrate when radio settings display is shown (two clicks)

### Reproduction

1. Switch on
2. Open Audio RX app and tune a station (I used FM)
3. Press the DFU button once to view the performance display/twice to view the radio settings display

### Expected behavior

1. The information is displayed
2. Audio playback stays the same

### Environment/versions

Version 2.0.0

gullradriel, 2024-03-16:
Hello.
It's not a bug, it's an hardware limitation.
Getting back counters is costing CPU cycle.
As for the scrolling area, it's scrolling all the content.
Use the level app and you can tune without DFU.

katherinehackworth, 2024-03-16:
Would the scrolling part not be a bug then though? It's presumably not supposed to do that but I don't know enough about how the waterfall is handled to know if it not doing that is possible

gullradriel, 2024-03-16:
Waterfall is scrolling the graphic memory itself.
Everything drawn on screen will by scrolled in the area, there is nothing yet we can do about it.

closed 2024-03-16 by gullradriel