portapack-mayhem/mayhem-firmware #1998: Microphone app crashing since the 14th March 2024.

John-Telek, 2024-03-16:
### Describe the bug.

The last 2 nightly releases have an issue with the microphone app. It crashes with the red screen of death or just hangs.  rel .14 and .16 are dud's but .13 was fine.

The Red Screen Of DEATH. Funnily enough, power cycle the unit and 9 out of 10 times it works the first time but crashes on exit occasionaly or throws up the red screen if you try to run the app again.

[image]

### Reproduction

Power cycle unit and try to use microphone app. It may hang or crash on exit. or if you can get out, using it again brings up the red screen.

### Expected behavior

It functions like expected.

### Environment/versions

14.03.24 and the latest 16.03.24

### Anything else?

Not at this stage. Still testing.

gullradriel, 2024-03-16:
Hello,
What's that version number ?

I can't reproduce on any of my 3 devices using the latest nightly.

There was a menu bug. Please update.

gullradriel, 2024-03-16:
Take the most recent from here: https://github.com/portapack-mayhem/mayhem-firmware/releases

Or use the web updater and flash the nightly : https://hackrf.app/

John-Telek, 2024-03-16:
Nightly Release - 2024-03-14
<https://github.com/portapack-mayhem/mayhem-firmware/releases/tag/nightly-tag-2024-03-14>
<https://github.com/portapack-mayhem/mayhem-firmware/releases/tag/nightly-tag-2024-03-14>
and Nightly Release - 2024-03-1
<https://github.com/portapack-mayhem/mayhem-firmware/releases/tag/nightly-tag-2024-03-14>
6
the ones before were ok

NotherNgineer, 2024-03-19:
@Brumi-2021 has reproduced this Guru fault too if the Microphone app is started when "RX audio" is enabled, and only if the Microphone app is run first.

I wasn't able to reproduce it exactly, but I see different issues:  The Microphone app hangs with my H2R4+ (after starting baseband audio but before painting the Microphone View screen).  Interestingly, I don't see any issue on my R9 version.

The memory allocation issue should theoretically be fixed in n_240316.  The most significant change in n_240314 is my audio DMA change, so I'll be investigating.

NotherNgineer, 2024-03-19:
The attached test firmware should resolve the issue:
[portapack-mayhem_dev.bin.zip](https://github.com/portapack-mayhem/mayhem-firmware/files/14645142/portapack-mayhem_dev.bin.zip)

closed 2024-03-19 by gullradriel

John-Telek, 2024-03-19:
> completed via #2011
> <portapack-mayhem/mayhem-firmware/issue/1998/issue_event/12163579945@
> github.com>

Great! The latest nightly release seems to be ok. Thanks again.

PR #2011 (2024-03-19): Stop audio DMA in baseband::shutdown()
Fixes a longstanding issue that the audio DMA engines, once started, continued to run freely after baseband::shutdown() was called.  It wasn't as much of an issue before PR's #1982 and #1987 because all background processes were starting both audio-input and audio-output DMA engines and letting them run forever.

This issue was causing the Microphone app to crash when switching between baseband processes that perform DMA operations in different directions.  

Fixes #1998

Test firmware available here:  https://discord.com/channels/719669764804444213/72210 [... cut]