portapack-mayhem/mayhem-firmware #2000: [Bounty Offered] FLEX pager message decoding

sa7mon, 2024-03-17:
### Description of the feature you're suggesting.

_This is basically a resurrection of #318 which was closed seemingly because bountysource went under._

I am offering $100 to anyone who gets FLEX message decoding for 2FSK (1600/3200 bps) and 4FSK merged into the [POCSAG app](https://github.com/portapack-mayhem/mayhem-firmware/wiki/POCSAG-Receiver).

I can provide sample recordings of messages from my area to aide in development.

Note:
> @ArjanOnwezen: As mentioned in https://github.com/portapack-mayhem/mayhem-firmware/issues/483. https://github.com/craigshelley/multimon-ng/blob/master/demod_flex.c might be a good start.

@notpike Are you interested in working on this? We chatted briefly on twitter a while ago but about doing this on the Flipper Zero. The HackRF platform is a much better tool for the job!

@micha-123 Do you want to match my bounty?

htotoo, 2024-03-18:
Maybe not in the near future, but when I'll finish my current project I may take a look into it.
But If anybody else wanna do it, feel free to do!

If you could attach some recordings here, that would help everyone. (be sure, if you play it back from PP you can receive and decode it with a computer before upload)

sa7mon, 2024-03-19:
> Maybe not in the near future, but when I'll finish my current project I may take a look into it. But If anybody else wanna do it, feel free to do!
>
> If you could attach some recordings here, that [...]

Yep, I'll get sample recorded and post a link here. Just gotta set up an SD card.

sa7mon, 2024-03-20:
Here is a small sample of a FLEX recording from today.

[sample.wav.zip](https://github.com/portapack-mayhem/mayhem-firmware/files/14661239/sample.wav.zip)

It can be decoded using `multimon-ng` as long as you re-sample from 48kHz (what the PortaPack records at) to 22.05kHz (what `multimon-ng` expects).

```sh
$ sox -t wav recording2.wav -r 22050 -t raw - | multimon-ng -t raw -a FLEX -
multimon-ng 1.3.0
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2023 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX FLEX_NEXT EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
Enabled demodulators: FLEX
FLEX: 2024-03-20 01:43:07 3200/4/A 04.093 [002051625] NUM 482
FLEX|2024-03-20 01:43:07|3200/4/K/C|04.100|4294961771|ALN|7120
FLEX: 2024-03-20 01:43:07 3200/4/C 04.100 [002051625] NUM 5 67
```

htotoo, 2024-03-20:
Thanks; maybe you got a c16 recording with it's txt part? I would like to replay it with portapack to an other portapack to test :)

sa7mon, 2024-03-20:
How do I create one of those? Still learning my way around this firmware.

kiwidotcom, 2024-08-10:
I 2 will pay 50 to anyone who gets this working correctly decoding flex pagers on the portapack

mattseabrook, 2025-03-23:
I’ve taken a look at the `demod_flex.c` from `multimon-ng` and it looks like a clean starting point for adapting FLEX decoding into the POCSAG app.

If integrated properly, it could support both 2FSK and 4FSK at 1600/3200bps using the same demod core, toggled via a config.

```c
// Example adaptation entry
if (mode == FLEX_3200_4FSK)
    decode_flex_4fsk(signal);
else
    decode_flex_2fsk(signal);
```
Let me know if there's still interest in this Bounty, please

sa7mon, 2025-03-23:
> Let me know if there's still interest in this Bounty, please

Yes, my bounty is still available. 

@kiwidotcom is your bounty still available?

stevenmcsorley, 2025-10-25:
> Here is a small sample of a FLEX recording from today.
> 
> [sample.wav.zip](https://github.com/portapack-mayhem/mayhem-firmware/files/14661239/sample.wav.zip)
> 
> It can be decoded using `multimon-ng` as long as you re-sample from 48kHz (what the PortaPack records at) to 22.05kHz (what `multimon-ng` expects).
> 
> $ sox -t wav recording2.wav -r 22050 -t raw - | multimon-ng -t raw -a FLEX -
> multimon-ng 1.3.0
>   (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
>   (C) 2012-2023 by Elias Oenal
> Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX FLEX_NEXT EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
> Enabled demodulators: FLEX
> FLEX: 2024-03-20 01:43:07 3200/4/A 04.093 [002051625] NUM 482
> FLEX|2024-03-20 01:43:07|3200/4/K/C|04.100|4294961771|ALN|7120
> FLEX: 2024-03-20 01:43:07 3200/4/C 04.100 [002051625] NUM 5 67

Confirming that FLEX decoding works perfectly using the current multimon-ng build.

Tested on Kali Linux (WSL2) using ( because Im using it for a web app in docker and Ive been testing kali out to make sure things work for my unholy purposes ):

`sox -t wav sample.wav -r 22050 -t raw - | multimon-ng -t raw -a FLEX -`

Output matches the expected 3200/4 FLEX frames — verified with the provided sample.

[image]

timelf123, 2025-12-03:
[image]

WIP, will post later

closed 2025-12-12 by gullradriel

htotoo, 2025-12-12:
And please feedback here if you got the bounty, for us to know

sa7mon, 2025-12-15:
> @sa7mon give the flex app a try and see if it checks the boxes for you!

On it! I haven't fired up the HackRF in a while so I'm getting everything set back up and hunting for POCSAG near me. Shouldn't take more than a day or so. 

If it all checks out, I'll email you regarding bounty payment. Appreciate the work!

timelf123, 2025-12-23:
Bounty was paid, thanks @sa7mon

Geofferey, 2026-11-10:
Here is my fork of the POCSAG pager for flipper zero, for the love of the game:

Even though this is closed, I think you will be interested. I'm posting it here because the mayhem flex app and the flex implementation I added to the POCSAG pager share some ancestry.

https://github.com/Geofferey/Flipper-Momentum-Apps/releases/download/flex-v2.0/flipper_pager.fap

It's capable of decoding FLEX 1600 (2FSK) and 3200/6400 (4FSK)... 

*FULL DISCLOSURE*: The code was entirely written using generative AI, however the ideas are completely mine. 

This is some serious voodoo going on with the CC1101 by sampling it's GDO0/GDO1 outputs directly. We are leveraging the Flipper Zero's MCU for software-assisted timing synchronization and symbol recovery to work round the radio's limitations. It's one of those things that'll make you genuinely go "WTF?!"... It is not 100% reliable decoding, but it's pretty damn good and a miracle the thing works at all 😂... 

[image]