portapack-mayhem/mayhem-firmware #2002: Adding a tutorial/wiki page on how to create .ppma apps

juanpedrosis, 2024-03-17:
### Description of the feature you're suggesting.

Would it be possible to have a guide or wiki page to explain how to code and compile the binaries to make custom apps for the hack rf. With the V2.0 and the ability to add apps directly to the SDCard this would be a nice thing to have.
Thanks

htotoo, 2024-03-18:
While it is not what you requested, it could help till there will be one: 
https://github.com/portapack-mayhem/mayhem-firmware/pull/1949/files
In this PR I created a simple ext app. (ignore the usb_serial_shell.cpp and messages.hpp and spi_image.hpp changes).

juanpedrosis, 2024-03-18:
Thank you, it will definitely help me

closed 2024-03-18 by juanpedrosis