portapack-mayhem/mayhem-firmware #2004: Pocsag not showing messages

m1ctk, 2024-03-17:
### Describe the bug.

All I get are short lines with F0 or F2 at the end. No actual messages as I do with PDW on the pc. I have the amp on and LNA at 40
Anyone having problems ?

### Reproduction

switch on go to pocsag app 

### Expected behavior

I was expecting full messages to be shown as on the windows PDW prog

### Anything else?

no

f1ghy, 2024-03-18:
Hi,
hummm, i try it, and i have also lot of pink "to many decode errors"... maybe need to try olders releases... interesting 🤔

f1ghy, 2024-03-18:
Hi,
hummm, i try it, and i have also lot of pink "to many decode errors"... maybe need to try olders releases... interesting 🤔

Hi, After a good night during a lotof pink "to many decode errors" PP is now verry slow. At the quick reboot, the splach screen come verry slowly... then after real OFF, then wait a bit, then ON all is fine !!! Memory Link ?

gullradriel, 2024-03-18:
Amp ON and LNA 40 looks a bit high. Maybe you are over saturating the decoder.
Check RxSaturation with DFU or Level app. It should be around 80% or less.

m1ctk, 2024-03-19:
OK I've got 24 32 now and the amp off and reception appeared to be better but I am still not getting messages

m1ctk, 2024-03-19:
OK so all good now, it was just a case of shifting the frequency 5 khz down, just wish you could scroll up with the main knob

NotherNgineer, 2024-03-20:
> OK so all good now, it was just a case of shifting the frequency 5 khz down, just wish you could scroll up with the main knob

BTW, if you'd like some change to the scrolling behavior, please clarify what you mean (and ideally submit an Enhancement request)

closed 2024-03-21 by NotherNgineer