portapack-mayhem/mayhem-firmware #2018: M0 Guru Meditation Hint: Hard Fault

Chriscassidy86, 2024-03-20:
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot
* Read: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/portapack-mayhem/mayhem-firmware/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/portapack-mayhem/mayhem-firmware/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2024-03-20:
Copied from title of duplicate PR #2019 from same submitter:

"M0 Guru Meditation Hint: Hard Fault, r0: 20001030, r1: 00070B9D, r2: 00007813, r3: 00000000, r12: 00000403, 1r: 000658AF, and pc: 00065316. Press DFU for Stack Dump im running the newest firmware. When i type in frequencies to load this happens every time."

Chris, which firmware version specifically?  Does this happen with nightly build n_240320?  Which app?

Chriscassidy86, 2024-03-21:
I'm running firmware v2.0.0.

NotherNgineer, 2024-03-21:
We need more details on exactly how to reproduce this.

ImDroided, 2024-03-26:
> We need more details on exactly how to reproduce this. Also please try the latest nightly build to see if it's already been fixed.

Not sure if its the same or not but I loaded n_240326 and for some reason the Analog TV app was giving a red out of memory screen after a few reboots it started working fine and has been since. no idea how to reproduce but it did eventually start working fine.

gullradriel, 2024-04-06:
It's happening on old versions. 
Make a clean install, reset all the settings.
See the wiki for the details (make a search on the page).
Please use the latest nightly or wait for next stable to come.
It only happens with bad settings and or bad sd cards.

closed 2024-04-06 by gullradriel