portapack-mayhem/mayhem-firmware #2022: `README.md` has link to `https://s.click.aliexpress.com/e/_Dkbqs2X` which does not work (anymore).

dreirund, 2024-03-21:
### Describe the bug.

The [`README.md`](https://github.com/portapack-mayhem/mayhem-firmware/blob/next/README.md?plain=1#L33) has a link to `https://s.click.aliexpress.com/e/_Dkbqs2X`, but this does not work anymore ("*Sorry, the page you requested can not be found:(*"):

```markdown
:heavy_check_mark: Another popular option is the clone of the [PortaPack H1](https://s.click.aliexpress.com/e/_Dkbqs2X).
```

Regards!

### Reproduction

1. Follow in the [`README.md`](https://github.com/portapack-mayhem/mayhem-firmware/blob/next/README.md?plain=1#L33) the link to [`https://s.click.aliexpress.com/e/_Dkbqs2X`](https://s.click.aliexpress.com/e/_Dkbqs2X),
2. see error page with text "*Sorry, the page you requested can not be found:(*".

### Expected behavior

Have a correctly working link.

gullradriel, 2024-03-23:
That's a job for @eried x-D

gullradriel, 2024-04-06:
Will be fixed by https://github.com/portapack-mayhem/mayhem-firmware/pull/2089

closed 2024-04-06 by zxkmm