portapack-mayhem/mayhem-firmware #2024: Cant open my replay in capture ?

Neomematrix, 2024-03-23:
### Describe the bug.

can someone help me, I cant open my capture folder. Also, do I delete old firmware that the portapack came with ? 

### Reproduction

wont open folder 

### Expected behavior

stuck 

### Anything else?

nope

gullradriel, 2024-03-23:
Maybe you have too much files in capture, and it's making the file manager going out of memory ?
Try reading your card on your computer.

Neomematrix, 2024-03-23:
Hi, I only have 6 files in captured. But I will get a bigger card today just in case. Any other ideas?

NotherNgineer, 2024-03-23:
(We're assuming that you mean that you can't open your captures in the Replay app.)

Can you open the CAPTURES folder in the File Manager app?

There's not enough information on the exact steps here.  In the Replay app, I assume that you're hitting the "+" and then pressing "down" key to select a C8 or C16 capture file and press "select"?

eried, 2024-04-02:
> Hi, I only have 6 files in captured. But I will get a bigger card today just in case. Any other ideas?

It can be just invalid data/sd (in Windows when you plug a "bad" sd it offers to fix it most of the time), does something happens when you try to open?

> Also, do I delete old firmware that the portapack came with ?

in the sd? yeah it is not used while using, only when flashing

gullradriel, 2024-04-06:
Use the latest nightly it can open a whole bunch of files in the same folder.
Will be usable in next stable too.

closed 2024-04-06 by gullradriel