portapack-mayhem/mayhem-firmware #2025: Cmake not rebuilding external apps when baseband code is changed

NotherNgineer, 2024-03-23:
### Describe the bug.

When a baseband module is changed that is used by an external app and firmware is rebuilt, the external app needs to be regenerated to include the updated baseband module.

### Reproduction

1. Modify a baseband module such as tone_gen.cpp.
2. Run "make"
3. Note that baseband modules are recompiled, but external apps were not rebuilt with the new baseband modules.

### Expected behavior

External apps need to be rebuilt if a baseband module is changed.

### Anything else?

Should be a simple makefile change but just documenting the issue right now.

gullradriel, 2024-03-23:
I got hit by it when trying your PR

NotherNgineer, 2024-03-24:
The makefile change in PR #2031 didn't work properly, so reopened this issue.

PR #2031 (2024-03-23): Add cmake dependency on baseband for external APPS
Cmake dependency fix:  If baseband code changes, rebuild external app ppma files since some external apps use baseband code.

Fixes #2025