portapack-mayhem/mayhem-firmware #2041: Fox Hunt

Hackrfan, 2024-03-26:
### Description of the feature you're suggesting.

Fox Hunt
Is there a description of this new function under Receiver?
Unfortunately I couldn't find it in the wiki.
Information is welcome.

gullradriel, 2024-03-26:
Not yet as it's still under development

gullradriel, 2024-03-26:
Poke @htotoo ^^

htotoo, 2024-03-26:
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Fox-Hunt

Maybe needs some polishing, bc of my english skills.. Feel free to fix it :)

Skipped the link to my repo from the wiki, don't want to advertise it this way, but if someone needs it: https://github.com/htotoo/ESP32-Portapack
(the ext module support esp code)

closed 2024-03-26 by htotoo