portapack-mayhem/mayhem-firmware #2043: Looking glass beep update blocks presets.

ImDroided, 2024-03-26:
### Description of the feature you're suggesting.

The new beep button that was added to looking glass takes up too much space and makes presets hard to see what you are on. Needs to be a check box or a new line added under the RES: VOL: line. There is blank black space above the waterfall below that line that would only need to be extended a small bit to fit another line without taking up much more of the waterfall.

NotherNgineer, 2024-03-26:
Adding @gullradriel

NotherNgineer, 2024-03-29:
I agree with the submitter that it would be nice to see more characters of the "Preset" description.  But of course we need to leave room above the waterfall for the red marker to be moved.

My suggestion would be to remove the "RANGE" field which is just displaying the difference between MIN & MAX frequency, and possibly put the beep setting there.

(I'm also wondering if perhaps the beep db setting should correspond to the signal level at the red marker versus an average across the whole range, but that's a different question.)

gullradriel, 2024-03-30:
Will be fixed on https://github.com/portapack-mayhem/mayhem-firmware/pull/2068 is merged

gullradriel, 2024-03-30:
I reduced the beep button syntax, and reduced PRESET: to P:

gullradriel, 2024-03-30:
> My suggestion would be to remove the "RANGE" field which is just displaying the difference between MIN & MAX frequency, and possibly put the beep setting there.

It's also used to lock the range or not. If you enable the range lock and then move min or max, you keep the range between them or not, hence I choose to leave it alone ;-)

closed 2024-03-30 by gullradriel