portapack-mayhem/mayhem-firmware #2046: Hackrf Broken DFU, black screen and all leds on

Davim09, 2024-03-27:
Hello
My hackrf bronken out of the blue, Before it was recognizing normally, now it stopped working, my pc don't recognize the dfu mode And Hackrf mode On windows and linux,

The driver.exe on windows don't open but I tried to install de .inf file, but nothing appers in devices manager.

I tried to upload a new firmware using the flash utility, but it didn't work and I got a black screen, the old firmware is 2.0.0 i tried to upload an older firmware 1.8.0, and after all the "uptade" from flash utility he gives me a black screen with all leds on

I tried to put it in DFU but no PC recognizes it I've already tried changing cables 3 times and change from pc, it is in hackrf the problem
[image]

What do I do?? Please help

zxkmm, 2024-03-27:
Hello, would you mind follow this guide and if you stuck, lemme know which step you stuck on?
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

Davim09, 2024-03-27:
> Hello, would you mind follow this guide and if you stuck, lemme know which step you stuck on? https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

Of couse i tried, the pc dont recognize the hackrf, i donr know if was the hardware or the firmware, i already change the pc

zxkmm, 2024-03-27:
Thanks. However you need to provide more information.
Are you sure you are on DFU mode?
driver.exe won’t open and it says what?
Have you tried Zadig?

Davim09, 2024-03-27:
> Thanks. However you need to provide more information. Are you sure you are on DFU mode? driver.exe won’t open and it says what? Have you tried Zadig?

yes i tried zadig, it shows nothing about hackrf one or dfu device.

zxkmm, 2024-03-27:
Try all the ENTER DFU MODE methods in this link and one of it maybe can bring you to DFU mode.
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

Davim09, 2024-03-27:
> Try all the ENTER DFU MODE methods in this link and one of it maybe can bring you to DFU mode. https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

I don't understand I didn't understand the last two options.

zxkmm, 2024-03-27:
Maybe try Google translate

Davim09, 2024-03-27:
> Maybe try Google translate

I am using right now, but how can it works if I put the cable after pressing the dfu button??

zxkmm, 2024-03-27:
I’m sorry I don’t understand your question.
Just follow the guide, try the methods one by one.

zxkmm, 2024-03-27:
And if you are having troubles reading the enter DFU methods, use google translate or ChatGPT to translate it.

Davim09, 2024-03-27:
> And if you are having troubles reading the enter DFU methods, use google translate or ChatGPT to translate it.

ok, i will try it

htotoo, 2024-03-27:
> > Maybe try Google translate
> 
> I am using right now, but how can it works if I put the cable after pressing the dfu button??

You HOLD the button while plug in the USB

Davim09, 2024-03-27:
yes i do this, i dont understant these 
[image]

zxkmm, 2024-03-27:
Knob: encoder, dial

Davim09, 2024-03-27:
> Knob: encoder, dial

oh ok, thanks

zxkmm, 2024-03-27:
You are welcome

Davim09, 2024-03-27:
> You are welcome

I've already connected my hackrf a few times to load it on my xioami source, could that be why the data transmission part of the hackrf's USB has burned out?? 
[image]
[image]
[image]

zxkmm, 2024-03-27:
It should output 5V1A before it successfully handshake, so it should be safe.
I didn't see anything burned out in your image tho.

htotoo, 2024-03-27:
Can you try with another computer?
And turn on sound, and check it there is any "new device connected" sound or not.

Davim09, 2024-03-27:
> Can you try with another computer? And turn on sound, and check it there is any "new device connected" sound or not.

I already tried 3 computers and nothing.

gullradriel, 2024-04-06:
Any news ? Got it working ?

Davim09, 2024-04-06:
> Any news ? Got it working ?

Nothing yet, try cleaning and nothing, I don't know if it's a problem with the hardware or the firmware, to recover the firmware I need to go into DFU mode and the computer recognizes it, but it goes into DFU mode But it doesn't recognize the USB, I did a continuity test with a multimeter and everything is fine going from the USB input to the NXP chip

gullradriel, 2024-04-06:
Are you using windows ?
Can you try on a linux ?

Davim09, 2024-04-06:
> Are you using windows ? Can you try on a linux ?

Yes, I tested it on 4 computers with Windows and 2 with Linux and nothing, nothing appears in the Linux lsusb and nothing in the dmsg

gullradriel, 2024-04-06:
Then it's your device/cables ?

Davim09, 2024-04-07:
> Então é o seu dispositivo/cabos?

No, it's a hackrf, I already changed the computer and the cables a lot of times, I don't know if it's a hardware problem or a firmware problem, before the black screen and all the LEDs were on, the hackrf was booting normally but it wasn't recognizing it via the USB, I went try uploading firmware via flash utility and BOOM! It would no longer boot and all the LEDs were on, and as it no longer recognized the USB there is no way for me to recover the firmware, is there any way for me to upload the firmware directly onto its chip?? I have a friend who understands this about firmware dumps

ToxxikHalo, 2024-04-07:
What exactly were you doing w your hackrf when this happened???

Davim09, 2024-04-07:
> O que exatamente você estava fazendo com seu hackrf quando isso aconteceu???

Trying to update the firmware using the flash utility, I had already done it once

ToxxikHalo, 2024-04-07:
Ok just making sure that was all...
How about taking the portapack off of the Hackrf and try connecting the Hackrf to a PC then... no portapack.

Davim09, 2024-04-07:
> Ok just making sure that was all... How about taking the portapack off of the Hackrf and try connecting the Hackrf to a PC then... no portapack.

I already did that and nothing, I cleaned contacts and nothing.

ToxxikHalo, 2024-04-07:
Oh ok, I didn't know if you actually split them apart... 
 I read about something similar months ago... I'll try to find it...
Man, at this point other than physical testing of components  it might be a lost cause unfortunately...  
Coulda been as simple as a static discharge or something like that...

Davim09, 2024-04-07:
> Ah, ok, eu não sabia se você realmente os separou... Eu li sobre algo parecido meses atrás... Vou tentar encontrar... Cara, neste ponto, além do teste físico de componentes, pode ser uma causa perdida, infelizmente... Poderia ter sido tão simples quanto uma descarga estática ou algo parecido...

It was out of nowhere that it stopped recognizing the USB, I don't know what could have happened, it always stays in a metal case from opensourcesdrlab

ToxxikHalo, 2024-04-07:
It was plugged in and sitting still or plugged in in your hand or being plugged in????

ToxxikHalo, 2024-04-07:
Do you know how to use a multimeter???

Davim09, 2024-04-07:
> Ele estava conectado e parado ou conectado na sua mão ou sendo conectado ????

I didn't understand, but I left it connected to charge, yes I know how to use a multimeter, I already did a continuity test and everything was fine, from the USB port to the NXP chip

ToxxikHalo, 2024-04-07:
Sorry, I'm asking if it was in your hand moving or was it sitting still when it went black??? I'm trying to just figure out if it's a physical problem or something wrong with the software side of things...

Davim09, 2024-04-07:
> Desculpe, estou perguntando se ele estava em sua mão se movendo ou estava parado quando escureceu??? Estou tentando descobrir se é um problema físico ou algo errado com o lado do software ...

From what I remember, I opened the .bin in the flash utility and left it on the table, it turned off and turned on again, but the screen was black and all the LEDs were on.

ToxxikHalo, 2024-04-07:
Oh ok gotcha...  
Yea ok to me that sounds more like an internal issue then...

Davim09, 2024-04-07:
> Ok... maybe get the schematic for HRf and do a more thorough test w multimeter...??? I know that's not an easy task but you've tried almost everything else...

My friend did this and everything went well

Uploading VID_20240406_115145.mp4…

Davim09, 2024-04-07:
> Ok... talvez pegue o esquema do HRf e faça um teste mais completo com o multímetro...??? Eu sei que não é uma tarefa fácil, mas você já tentou quase todo o resto...

My friend who works with communication radios and does maintenance did this and everything worked fine.

https://github.com/portapack-mayhem/mayhem-firmware/assets/143907552/363604d7-4f40-453d-b4d4-0a19ae0a32a9

ToxxikHalo, 2024-04-07:
Ok gotcha...  yea based on how it happened I'm thinking somehow the flash went wrong or something wierd...

How about your SD card... have you tried a different card?

Davim09, 2024-04-07:
> Ok, entendi... sim, com base em como aconteceu, estou pensando que de alguma forma o flash deu errado ou algo estranho...
> 
> E quanto ao seu cartão SD... você tentou um cartão diferente?

It's not the same since I bought HACKRF, and the flash utility worked before

ToxxikHalo, 2024-04-07:
Yea I was thinking maybe change it again but it can't be that anyway cause you tried the HRF separately already....

ToxxikHalo, 2024-04-07:
I'm trying to associate how a portapack fw update could damage the hackrf itself and bad enough that it can't even directly connect to a PC as designed....  
I'm gonnatake a little while and read up on the HRf build and it's program pathways as best I can. I'll let you know if I find anything that might help...🤷

Davim09, 2024-04-07:
> Ok gotcha... yea based on how it happened I'm thinking somehow the flash went wrong or something wierd...
> 
> How about your SD card... have you tried a different card?

I also don't know what could have happened, do you know if there is any way for me to reprogram the firmware directly on the HackRF chip?

Davim09, 2024-04-07:
> Estou tentando associar como uma atualização do portapack fw pode danificar o próprio hackrf e é ruim o suficiente para que ele não consiga se conectar diretamente a um PC conforme projetado .... Vou demorar um pouco e ler sobre a compilação do HRf e são os caminhos do programa da melhor maneira que posso. Avisarei se encontrar algo que possa ajudar...🤷

Ok Obrigado!!

ToxxikHalo, 2024-04-07:
Just to clarify... so you plug it in and the LEDs turn on and screen stays black??? And that's it?
Nothing at all happens that you notice?

Davim09, 2024-04-07:
> Só para esclarecer... então você conecta e os LEDs acendem e a tela fica preta??? E é isso? Nada acontece que você perceba?

Yes, Look at the video 

https://github.com/portapack-mayhem/mayhem-firmware/assets/143907552/a92c896d-428d-4a8f-821b-8153d7f91273

ToxxikHalo, 2024-04-07:
Ok gotcha... I'm reading over the HRf docs now so hopefully there's something in there to help... I'll let u know. If you have time you might wanna give them a look also... not the mayhem docs, the HRf docs I mean...

Davim09, 2024-04-07:
> Ok, entendi... Estou lendo os documentos do HRf agora, então espero que haja algo lá para ajudar... Avisarei você. Se você tiver tempo, talvez queira dar uma olhada neles também... não nos documentos do caos, quero dizer, nos documentos do HRf...

Thanks! Yes, I'll look, but apparently I'm the only person in the world who has this type of problem.

ToxxikHalo, 2024-04-07:
Lol someone has to be the first... Hopefully first to fix it too.

ToxxikHalo, 2024-04-07:
I know you said you tried DFU reset already. But have you tried it while
the HRf was separated from the Portapack too???
And try both methods too...
 hold the DFU button and press and release the reset button then release
DFU button.
And
Hold DFU button and power on the HRf then release DFU...

I know you've done all the basic stuff already and you seem to know what
your doing, I just wanna cover all the bases just in case...

ToxxikHalo, 2024-04-07:
https://hackrf.readthedocs.io/en/latest/updating_firmware.html

Some more options to try....

Davim09, 2024-04-07:
> Eu sei que você disse que já tentou redefinir o DFU. Mas você já experimentou enquanto o HRf estava separado do Portapack também??? E tente os dois métodos também... segure o botão DFU e pressione e solte o botão reset e depois solte o botão DFU. E segure o botão DFU e ligue o HRf e solte o DFU ... Eu sei que você já fez todas as coisas básicas e parece saber o que está fazendo, só quero cobrir todas as bases para o caso ...
> […](#)

yes i already to all the methods to DFU mode and nothing in windows and linux

Davim09, 2024-04-07:
> https://hackrf.readthedocs.io/en/latest/updating_firmware.html Mais algumas opções para tentar....
> […](#)

I was interested in this part, what is a Jawbreaker?? do you know if has a video instructing how to do it??
[image]

ToxxikHalo, 2024-04-07:
> https://hackrf.readthedocs.io/en/latest/updating_firmware.html Mais
> algumas opções para tentar....
> … <#m_548348617565829536_>

Jawbreaker is/was the beta version or prototype of the hackrf one... all I
can tell you is that as far as I know the HRf is the better SDR.
Depending on what you wanna do you can also combine multiple HRf together
to make a full duplex sdr or even more of them if needed that can be
triggered from a single HRf.
Then there'd the opera cake board also to check out...

Davim09, 2024-04-07:
> Jawbreaker is/was the beta version or prototype of the hackrf one... all I can tell you is that as far as I know the HRf is the better SDR. Depending on what you wanna do you can also combine [...]

Gotcha, Seems to me to be a physical issue with the micro USB port on it, because before the black screen and all leds on the HackRF was working perfectly

Davim09, 2024-06-11:
Hi! new update, the OpenSourceSDR store send me a new HackRF board, i send my issue for they via discord and telegram, thanks so much!
[image]

Davim09, 2024-06-11:
Hi! new update, the OpenSourceSDR store send me a new HackRF board, i send my issue for they via discord and telegram, thanks so much! the old board i will send to a microsolder friend to try to fix it.

[image]

htotoo, 2024-06-11:
This is why OSDR is the recommended store ;)

closed 2024-06-11 by htotoo