portapack-mayhem/mayhem-firmware #2053: Add beep option to other apps

NotherNgineer, 2024-03-28:
### Description of the feature you're suggesting.

Other apps could use an optional beep when a packet is received, such as Pocsag, ERT, Weather, and TPMS.

NotherNgineer, 2024-03-28:
I wonder if this option should be on the title bar.

NotherNgineer, 2024-03-28:
I agree that having it on the title bar probably isn't necessary since we already have a mute button on the title bar and that should be enough for most apps.

But do you think we need the "beep on packet" option to be configurable for each app separately, or is it sufficient to have a single "beep on packet" setting stored in pmem that applies to every app?

htotoo, 2024-03-28:
Beep on packet settings per app would take a lot screen space i think, and can't afford that on every app (a lot is crowded already)
I think the "global" setting, with the speaker title bar icon would be enough.

NotherNgineer, 2024-03-28:
In PR #2058, I'm proposing to modify the Settings -> Audio screen to add this, per feedback above.

NotherNgineer, 2024-03-29:
I was going to add support to SubGHzD too, but there isn't space for a larger baseband image currently.

I'm going to mark this one as completed now.

gullradriel, 2024-03-29:
Nice work