portapack-mayhem/mayhem-firmware #2057: Configurable Root folder idea

NotherNgineer, 2024-03-28:
### Description of the feature you're suggesting.

Perhaps the root folder could be configurable in Settings (default would be "/" as it is now) and the setting could be stored in a file in the _real_ root of the SD card.  That way multiple SD card images could be placed on the SD card and the user could switch between them.  Before the user downloads a new SD card image, they could rename the old folder to make sure that no user files are overwritten.

This setting would apply to all apps, but File Manager should still be able to see the entire SD card contents, so that a user could browse and pick a file from anywhere on the SD card, and move folders between them.

This could perhaps be a possible alternative to PR #1967 and PR #1968 (which were meant to prevent user files from getting overwritten when a new SD card image was downloaded).

Of course older firmware wouldn't support this and would always assume that the root directory was "/" after a firmware downgrade, but there would be no incompatibility issue with firmware _upgrades_, unlike in the PR's mentioned above.

(This could additionally be helpful for users that incorrectly downloaded their files in an "SDCARD" folder instead of to the root; they could simply configure the root folder name to match whatever folder they unzipped the contents into.)

htotoo, 2024-03-28:
zxkmm's solution seems easier by the users perspective, and from the devs (eg updater) too.
I know that would require a big change on the first version update, but would be easier to manage in the long run.

NotherNgineer, 2024-03-28:
Withdrawing suggestion

closed 2024-03-28 by NotherNgineer