portapack-mayhem/mayhem-firmware #2059: BLE RX/TX only working when charging or TCXO is disabled

rwardd, 2024-03-29:
### Describe the bug.

I have recently purchased a PortaPack H2 from OpenSourceSDRLabs, flashed it with the lastest firmware, added the required SD card loaded with the latest application release. When using BLE RX mode, the device does not receive any packets, unless it is either plugged in (charging), or the 10 MHz TCXO is disabled. The same happens on TX/BLE spam - nothing gets sent when the device's TCXO is enabled. It can receive AIS, audio etc when the TCXO is on, it just seems to be BLE that isn't receiving anything. I have updated the CPLD too.

### Reproduction

1. Turn device on
2. Navigate to Receive -> BLE RX
3. No packets are received when TCXO is enabled and the device is unplugged

### Expected behavior

1. Would expect the BLE Rx to start receiving packets (there is BLE activity in the vicinity)

### Environment/versions

Mayhem firmware v2.0.0, r4 H2 PortaPack, r9 HackRF.

NotherNgineer, 2024-03-29:
What is the charge level on the battery (how many blue LEDs are on when unplugged)?  I think I've heard mention of some TXCO's not working properly when the battery is very low.

rwardd, 2024-03-30:
@NotherNgineer Battery is full/near full (all 4 LEDs are lit) when unplugged.

NotherNgineer, 2024-03-30:
I'm guessing it's a hardware issue (TCXO may be requiring more voltage than the regulator is delivering).

You might want to try running the Looking Glass app with preset BL/WIFI 2.4GHz in F-LEVEL-V mode and compare what you see with TCXO on versus off.

Perhaps someone else will have an idea.

rwardd, 2024-03-31:
I agree I think that might be the case too. Will try the Looking Glass test, thanks for the suggestion. I have contacted the manufacturer, and it seems like others might be experiencing something similar.

gullradriel, 2024-04-06:
Which TXCO are you talking about ?
The internal should be disabled, and the external one enabled.

gullradriel, 2024-04-10:
> The same happens on TX/BLE spam - nothing gets sent when the device's TCXO is enabled.

As it can not be clear about which settings you're talking (internal or external one ?) I can give you a simple thing to check:
In Settings/Radio, nothing should be checked and it should use external 10MHz TXCO.

If it's not working properly except when plugged in, then your TXCO must be one of those who need more power to function properly. 

Anyway it's not a Mayhem/Firmware bug.

I can confirm it's working all good on my 3 devices.

gullradriel, 2024-10-12:
Closed due to inactivity

closed 2024-10-12 by gullradriel