portapack-mayhem/mayhem-firmware #2060: 2.0.0 runs once and then error after first reset.

pcpackrat, 2024-03-29:
I loaded 2.0.0 and copied files to the SDCard.  Hackrf loads and apps work. 
If I reset the device, I get a red box screen with HACKRF CPLD FAILED.  

To get it running again, I placed the unit in DFU, run defu_hack_rf_one.bat and then flash_portapack_mayhem.bat and it loads 1 more time until a reset.

Rolled back to 1.9.1 and hackrf is operating fine.

I copied files over the top of what was on the SDCARD from 1.9.1

gullradriel, 2024-03-29:
@bernd-herzog any idea ?

deepuj, 2024-03-29:
I am seeing something similar. Occasionally the device goes into HACK RF CPLD failed; reset button does not help much.  If we wait for the device to completely drain the Li-ion battery (or disconnect it), it comes back up (I do have the button cell installed; but did not have to take it out for this).

gullradriel, 2024-04-02:
Maybe try to disable config mode in settings. It's enabling a special boot mode that is sometimes triggered by some devices at init.

pcpackrat, 2024-04-03:
Is that on the Sd card?

-Michael

pcpackrat, 2024-04-03:
Disabling config mode from Settings -> Config Mode seems to cleared my issue. Thanks,

closed 2024-04-03 by pcpackrat