portapack-mayhem/mayhem-firmware #2071: Fileman: paginate instead of keeping a list of all elements in directory

gullradriel, 2024-03-30:
### Description of the feature you're suggesting.

Actually Fileman keeps a list of the whole directory in memory and literally eat the memory.
We want to only keep in memory what's on screen, and even if possible only do set_text or equal on the needed lines instead of loading a list (I mean use set_text on them as we parse the linked list)

NotherNgineer, 2024-03-30:
Another thing I'd like to see in Fileman is the ability to sort files by name or time stamp (this is probably even more unlikely if we only load a partial list into memory, but I thought perhaps the design could take that into consideration)

NotherNgineer, 2024-03-30:
Just a thought:  I think we set aside about 70KB of RAM space for baseband, and I wonder if we could either allocate space for the file list from that, and/or move part of the File Manager code to baseband.  Of course this assumes baseband doesn't need to be running when selecting files.

NotherNgineer, 2024-03-30:
Using DFU, it also appears to me that File Manager isn't completely deallocating space for previously-opened folders until File Manager is exited.

Example:  When I start File Manager from the utilities menu, there's about 25KB free (M0 core + M0 heap) when looking at the root directory.  But if I open _and close_ my SCREENSHOTS folder with 95 files in it, available memory has dropped to <18KB while viewing the root directory.  The missing >7KB seems to be deallocated when File Manager is exited, but perhaps we could look into deallocating this space when the view of the large folder is exited.  Otherwise it seems that I can get intermittent Guru "out of memory" faults when navigating between folders.

gullradriel, 2024-04-02:
- at PP start: update_all_fileman_db
- at any time we need to read a directory content and navigate/choose a file: open target/.fileman.db and use it. When clicking on a directory, open directory/.fileman.db and so on until a file is selected
- display is using 8 menu entries ( or 8 whatever will be used to represent one line) in Fileman (instead of the number of files in the directory + the included directories)

gullradriel, 2024-04-02:
@NotherNgineer I'm editing the OP to reflect what I thing we need now

gullradriel, 2024-04-02:
Solved by #2077

closed 2024-04-02 by gullradriel