portapack-mayhem/mayhem-firmware #2080: BLE Spam and BLE Rx wont work when not plugged into Charging station

BatmanWhoEats, 2024-04-03:
### Describe the bug.

As i said the portapack wont receive anything on ble rx or transmitt anything on ble spam when not being plugged into the charging station(altough its fully charged)but literally the second i plug it in as shown in the video it magically starts working 

https://github.com/portapack-mayhem/mayhem-firmware/assets/165921193/249c1ec7-fb14-40c9-8e6c-1a3b0f30cd66

### Reproduction

1.go to Ble Rx
2.have nothing plugged in(and receive nothing
3.plug it into an charging station
4.see how the packets start flodding in as soon as the usb enters

### Expected behavior

It should just work normally without being plugged into an charging station

### Environment/versions

hackrf porta pack h2(from OpenSourceSdr Lab) with mayhem 2.0.0

### Anything else?

There isnt any problem with anything else. Soundbrd works just fine,looking glass and anything in that low frequency range that i tested but anything with 2.4ghz or higher wont work without it being plugged into the charging station

NotherNgineer, 2024-04-03:
Sounds like the same issue as #2059

gullradriel, 2024-04-06:
Duplicate of https://github.com/portapack-mayhem/mayhem-firmware/issues/2059
Please comment on that one. Thanks.

closed 2024-04-06 by gullradriel