portapack-mayhem/mayhem-firmware #2094: SWR meter

amelillo, 2024-04-07:
### Description of the feature you're suggesting.

Hello,
just wondering: would it be possible to measure SWR readings for the antenna attached to the HackRF on a specific frequency?
I don't know if the HackRF chip or the Portapack one can provide SWR readings but, as we can receive and transmit in a huge frequency range, it would be great to be able and test our specific antennas to fulfil our needs at their best.
Then - but I think we shouldn't worry too much about it, as tx power is very low - we could safely tx withouth the risk of burning the final RF transistors.

ToxxikHalo, 2024-04-07:
Similar to a VNA??? That'd be a great addition if it was possible for
sure... being able to tune antennas and pair one for a specific use would
be incredibly useful for HRf...

gullradriel, 2024-04-08:
All available metrics are already exposed,
and I don't think we can measure SWR from within the portapack.

closed 2024-04-08 by gullradriel