portapack-mayhem/mayhem-firmware #2097: Prizim

Carreker, 2024-04-09:
### Description of the feature you're suggesting.

Hello,

I would like to have a feature on the hackrf that I believe could be done and it also would add a ton of features and make this Mayhem project even better!  I am not sure of the legality of it but it would be really cool if we could find a way to add the palm pilot prizim OS of the 1990's to the Mayhem project.  I remember having a gps radio that I could attach to the palm pilot and for some reason I was able to view people as heat signatures on it.  I am interested in what it takes to make something like this happen I would like to be involved if this ever sees the light of day!  Thank you for your time and consideration!!

eried, 2024-04-10:
[Fork](https://github.com/portapack-mayhem/mayhem-firmware/wiki/How-to-collaborate) the repo, start thinkering with your idea and link it to this issue. That would be the way to kickstart something :)

I do not believe is technically possible, but it is a good start for experimenting

eried, 2024-04-10:
I will close this issue for now, but feel free to reopen

closed 2024-04-10 by eried