portapack-mayhem/mayhem-firmware #2103: GPS Sim wont work

HHelpMelol, 2024-04-12:
### Describe the bug.

hey so ive been trying the gps sim a couple times now but it doesent seem to work. i did exactly that what the wiki said and also got the latest file from https://cddis.nasa.gov/archive/gnss/data/daily/2023/brdc/ 

### Reproduction

1.gps-sdr-sim -e  brdc3650.23n -l 30.036028807598665, 31.320475630204022, 100 -b 8 -o EGYPT.C8
2. create a txt with sample_rate=2600000
center_frequency=1575420000
3. put both of the files into the gps folder on the hackrf portaback
4. go to gps sim app on the hackrf
5. open the EGYPT.C8 File
6. start 

### Expected behavior

1.it should show me that iam at the coordinate (in my case egypt) on googlemaps 

### Environment/versions

Mayhem v2.0.1

NotherNgineer, 2024-04-13:
If that's the actual command line in step 1 above, you should remove the space characters after the commas.

HHelpMelol, 2024-04-13:
> If that's the actual command line in step 1 above, you should remove the space characters after the commas.

just tried it but still wont work

HHelpMelol, 2024-04-13:
> If that's the actual command line in step 1 above, you should remove the space characters after the commas.

i also got the newest bdrc file because before i had the 2023 one but it wont work

eried, 2024-04-16:
Have you read the wiki in detail? location could be retrieved from different sources by some devices, and some wont "believe" the radio position is correct if there is something weird

gullradriel, 2024-04-16:
Please thoroughly check https://github.com/portapack-mayhem/mayhem-firmware/wiki/GPS-Sim#compatibility

eried, 2024-05-05:
I will close this for the moment, since it seems was not a bug

closed 2024-05-05 by eried