portapack-mayhem/mayhem-firmware #2104: Power lock button (not an issue)

D0lby55, 2024-04-13:
### Description of the feature you're suggesting.

This is a "Power button" with a ring lock! with this a PPH2 can travel in you're pocket! you can lock the power button by unscrew the ring! there two model Normal and Mayhem!

[image]
[image]
[image]
[button.zip](https://github.com/portapack-mayhem/mayhem-firmware/files/14968136/button.zip)

NotherNgineer, 2024-04-13:
IMO, I don't think Github Mayhem **Firmware Issues**  is the correct place to post this.  I would suggest moving this discussion to the Discord channel, or to some 3-D design site such as Thingiverse.

The power button on the H2 is of course the rotary encoder.  It doesn't look like it'd be easy to spin the encoder dial on either version of this, especially without rotating the ring lock.

D0lby55, 2024-04-13:
sorry you can remove it, that's only to lock the push button you could alway's spin it!

zxkmm, 2024-04-14:
Thanks for sharing! You can share it in this Discord server: https://discord.gg/PeNJBAN3
If you don't mind, I'm going to close this ticket (since it's not an issue)
But if you want some discussions based on this topic, feel free to re-open it.

closed 2024-04-14 by zxkmm