portapack-mayhem/mayhem-firmware #2105: RX SubGhzD Step adjust

Majorstoned, 2024-04-15:
### Description of the feature you're suggesting.

Is it possible to write the Option to adjust the Steping for SubGhzD RX. 100khz is very large for testing following Frequencys like 433.920 -> 433.930 ... thanks 73

Majorstoned, 2024-04-16:
Can i via a simple Way the new Apps direct send to the Portapack, or Change the issous direkt on the Pack? Without wait for the next Update.

eried, 2024-04-16:
@Majorstoned You can use a nightly or follow the wiki instructions to compile

zxkmm, 2024-04-16:
@eried no they want set step in subghzd app instead of record app

htotoo, 2024-04-16:
Go to discord, and there is an update file in the #test-drive chan

zxkmm, 2024-04-16:
No you actually didn’t make the step adjustable. They want the step adjustable like capture app. Unless I didn’t read your new changes?

zxkmm, 2024-04-16:
Oh wait I read it wrong somehow

zxkmm, 2024-04-16:
Sorry @htotoo @eried

closed 2024-04-16 by zxkmm

Majorstoned, 2024-04-16:
Works well via Discord, but wefax ppma is outdated... im happy 10khz is very fine

PR #2106 (2024-04-16): Smaller freq changes with encoder on SubghzD and Weather Station
Resolves #2105