portapack-mayhem/mayhem-firmware #2107: AIS RX - Monitor both freqs at the same time

tubalainen, 2024-04-16:
### Description of the feature you're suggesting.

Does the CPU have enough horsepower to monitor/RX both the VHF AIS frequencies (87B and 88B) at the same time? Would love to have that option next to monitor each freq individually. Thanks to all the devs for a fantastic firmware!

eried, 2024-04-16:
Do you mean channel A and B or decoding those two message types (87B and 88B)?

tubalainen, 2024-04-16:
> Do you mean channel A and B or decoding those two message types (87B and 88B)?

Nope.
I mean decoding both the channels 87B and 88B at the same time. Not the the simplex A channels/freqs. :)
Not sure what you mean with "two message types", message types to me for IAS refers to the TU M.1371-5 standard (27 message types in that standard).

Perhaps I am misunderstanding you question...?

eried, 2024-04-16:
oh, so `87B (161.975 MHz) and 88B (162.025 MHz)`? I though it was called A/B

Im thinking how cheap equipment does it, like an rtlusb dongle, I imagine most modern receivers tune to both without switching

tubalainen, 2024-04-16:
> oh, so `87B (161.975 MHz) and 88B (162.025 MHz)`? I though it was called A/B
> 
> Im thinking how cheap equipment does it, like an rtlusb dongle, I imagine most modern receivers tune to both without switching

Yep, exactly! There´s lots of projects out there that does exactly this. Monitors and decodes both 87B (161.975 MHz) and 88B (162.025 MHz) at the same time.
I personally use this running in a docker here at home: https://github.com/jvde-github/AIS-catcher I use an RTL-SDR on it. :)

TL:DR - The bandwidth is very low hence the two frequencies are so close to each other, so the demodulation and decoding capacity should(?) be there in the portapack?

eried, 2024-04-16:
yeah it should be possible, eventually losing some messages, but thats a given. If you feel adventurous to try to hack a solution I would start with the jammer code that already offers the freq hop 😁

tubalainen, 2024-04-16:
> yeah it should be possible, eventually losing some messages, but thats a given. If you feel adventurous to try to hack a solution I would start with the jammer code that already offers the freq hop 😁

Thats the good thing here - no hopping required. We can monitor both freqs at the same time hence they are so close to each other in freq.

NotherNgineer, 2024-04-17:
There are no ships in my area to test this, but since we're setting the RX bandwidth to 1.75MHz it's likely that we're _already_ picking up broadcasts on both frequencies, since they're so close, regardless of the channel setting.

tubalainen, 2024-04-17:
> There are no ships in my area to test this, but since we're setting the RX bandwidth to 1.75MHz it's likely that we're _already_ picking up broadcasts on both frequencies, since they're so close, regardless of the channel setting.

Hi N! Yea, that might be the case. But it only decodes one channel at a time. Perhaps this is a very simple adjustment of the code to bring in a great feature? - Decode both channels at the same time. In the "Ch" selection, one can pick both, 87B or 88B. :)

NotherNgineer, 2024-04-17:
As far as I can tell, changing the channel only modifies the receive frequency (I don't see any decoding changes when the Ch field is changed).  So my guess is that this should already work.

Brumi-2021, 2024-04-17:
Hi , @NotherNgineer ,
As far as i remeber , my guess is that   it will not work . I think that  we are prefiltering that signal with a narrow filter to allow only one channel Content . (As it is intended ) 
I can easily check it and report  here, but that is my understanding.

I tested what you said in GNUradio and other SDR's receivers, and it works quite well .
I think we could consider to add that double simultaneous both channel  decoding option . (But Not sure if it will skip  some small % message if they both are overlapped in time ) 
I will update that comment with my next quick dual test .

Cheers

Brumi-2021, 2024-04-18:
Hi , @tubalainen , concept wise , I know how to do it by gnuradio or any other simulation, but I need to learn and think how to implement in Mayhem  . After some other current Job i will try to consider about it , but if  any other colleague wants to start now , you are also welcome !

NotherNgineer, 2024-04-21:
If we really can monitor multiple frequencies at once, it could also help in some other apps including TPMS and BLE-RX.

NotherNgineer, 2024-05-15:
I was near a marine harbor this week and it looked like the same ships appeared in the list regardless of the channel setting.

przemo-c, 2024-08-05:
> I was near a marine harbor this week and it looked like the same ships appeared in the list regardless of the channel setting.

I'm not sure that's verification that it decodes both. Almost all ships and other AIS transponders sent messages on both channels when I go via rtl-sdr and ais-catcher on android.