portapack-mayhem/mayhem-firmware #2108: RTL_433 like tool

gomond, 2024-04-16:
### Description of the feature you're suggesting.

It would be awesome if you could add something like RTL_433 as I use it a lot on linux. but would be awesome in a portable version.

htotoo, 2024-04-16:
If anyone want to work on this, here is a little help: 
https://github.com/NorthernMan54/rtl_433_ESP/ 
This uses the cc1101 for the timings, so with the SubghzD part on PP this could be portable a bit easier, than the original rtl433.
Also I don't know would it fit the baseband size limit..

NotherNgineer, 2024-04-18:
It seems that lots of people would be interested in this enhancement; see #113 (apparently closed due to inactivity) and #254.

ImDroided, 2024-04-19:
This would be a good addition.

ggenny, 2024-07-14:
a basic tool that, with the appropriate parameters, captures ASK sequences would be helpful. I currently use a gnuradio pipeline, but an integrated solution would be better. SubghzD misses too many signals; it's too 'minimal', we really need something more straightforward. Unfortunately, I don't clearly understand the app development process, otherwise I would have integrated it myself

mautas, 2025-07-23:
I would find this a great enhancement to Portapack