portapack-mayhem/mayhem-firmware #2112: mayhem 2.0.1 looking glass full lock up

oj-briodd, 2024-04-18:
### Describe the bug.

i just built my portapack and flashed it with the newest stable release 2.0.1 but some apps primarily looking glass crashes upon opening or when i change any settings and locks up the whole system so i either have to press the reset button or hard boot the system

### Reproduction

i don't know

### Expected behavior

looking glass opens / works

### Environment/versions

2.0.1 and 24/04/18 nightly

eried, 2024-04-24:
Does Capture work? When it locks up, is the screen frozen or just black?

gullradriel, 2024-04-24:
Already answered a similar problem.
You should check/update your Sdcard contents.
Particularly Looking Glass presets file.
If you messed it up, you're good for a lockup.

gullradriel, 2024-04-24:
Anyway the issue is on your side/device.
I can not reproduce it on my devices, and I have 3 different ones.

eried, 2024-04-25:
Yeah, lets close this for now, @oj-briodd can reopen if he has more details.

closed 2024-04-25 by eried