portapack-mayhem/mayhem-firmware #2117: Monitor full spectrum 1MHz to 6GHz  in Portapack with "Hackrf_sweep" "QspectrumAnalyzer"

TropicalPalmtree, 2024-04-21:
### Description of the feature you're suggesting.

HackRF connected to a PC can show a waterfall of the full spectrum 1MHz to 6GHz, using "QSpectrumAnalyzer" 
https://github.com/xmikos/qspectrumanalyzer
It is based on the "hackrf_sweep" tool: 
https://github.com/greatscottgadgets/hackrf/blob/master/host/hackrf-tools/src/hackrf_sweep.c

Is it possible to run the QSpectrumAnalyzer on the Portapack (Mayhem H2) ???
Or build a similar app using hackrf_sweep, for the future Mayhem firmware ???

Thansk ;)

gullradriel, 2024-04-24:
Not possible in portapack mode, there is simply not enough CPU power to process it.

gullradriel, 2024-04-24:
Use your portapack in hackrf mode and use a computer.

closed 2024-04-24 by gullradriel