portapack-mayhem/mayhem-firmware #2120: Portapack automatically deleting all files from sd card on boot.

Yabonbaton, 2024-04-25:
### What happened?

I do everything right, I format the SD card in fat32 format, doing a full formatting, throw the firmware files there with the card, take out the SD card, put it in the portapak, plug it in. My screen hangs for a while, everything is deleted except adsb map and that's it. I delete all the applications, in general, all the necessary things. Firmware version 2.0.1. I tried to download applications through hackrf.app, but in the end, they are also deleted after some time, 30-60 seconds. I have no idea what the problem could be, tried several SD cards, all to no avail.

### Your hardware

Hackrf one Portapack h2, mayhem 2.0.1

### Tests

- [ ] I managed to update/boot previously

gullradriel, 2024-04-26:
You may have buggy card that are circularly writing datas.
There is no such thing in the firm as 'delete files on boot'.
You may also check in Settings/Sdcard that the 'high speed model is disabled.

NotherNgineer, 2024-04-26:
Also please make sure that you're safely removing the SD card from your computer.  PC operating systems normally cache SD card writes and the write operations may not be complete if you remove it without first telling the OS that you plan to remove it, and then waiting for the message to pop up saying it's safe to do so.

(I mention this since you didn't list this step above, and I have no idea of your computer skill level.)

ZetaWaves, 2024-09-26:
The issue is likely that you have a fake 4GB SD card.
Many of these are going around in China. They're actually 1GB SD cards set to display 4GB. 

What's happening is you are overwriting your own data, it becomes corrupted, and then it is removed on disconnect.

Advice: Get a brand new, reputable SD card that is 16GB - 32GB in size. Format as FAT32, load files, good to go.

zxkmm, 2024-10-06:
Close as inactive from OP for months and looks like a very rare case or user fault.

closed 2024-10-06 by zxkmm