portapack-mayhem/mayhem-firmware #2127: How can I make sure my HackRF is working properly?

Naster17, 2024-04-27:
### Describe the bug.

I'm afraid that someone might have touched the optical antenna with their hand while my hackrf was working. And so this thought does not leave me, how can I check my hackrf

### Reproduction

-

### Expected behavior

-

zxkmm, 2024-04-28:
don't worry much. it's not like definitely would happen thing. 

when i wrote that page, it's more like a "try to prevent" list instead of "extremely dangers" list.
If the physical touch didn't released strong static electricity, and didn't bring much impedance changing, it's *usually* safe (but still, try to prevent)

so i edited the wiki again. and answered your question.

closed 2024-05-11 by zxkmm

zxkmm, 2024-05-15:
You are welcome!