portapack-mayhem/mayhem-firmware #2134: Tetra signal detection

brisioksss, 2024-04-30:
### Description of the feature you're suggesting.

It would be awesome to have an app that notifies (with a beep) when specific signal power rises

In this application it is used to notify nearby ambulance/firefighters/police vehicles 
https://www.youtube.com/watch?v=mKjc3Z_6IXY

Basically it could be as an addition to the 'level' app, only need to configure thresholds for different sounds 
Thank you

Radiomix2000, 2024-05-01:
Technically for  that we need to calculate spectral density in known freq band. All Tetra mobile radios in TMO mode periodically send the signals on uplink freq (380-385). Different sync signals and frequency correction bursts (actually CW carrier very short in time). No need to decode tetra constellation for that. Only the basic FFT calculations.

brisioksss, 2024-05-01:
Really interesting! It would be an amazing addition to this great hardware. 
I am sure a lot of people would be interested in this

schmiernippel, 2024-05-03:
Such an app would be wunderful 🤗🤗🤗

brisioksss, 2024-05-17:
A python code that does exactly this.
Need to find a brave soul to port this over :^)

https://github.com/NukeAlchemist/blupi/blob/master/blupi-scan.py

JAKAMI99, 2024-11-07:
I also would love this, a Target Blueye costs over 1000 USD, while I am sure, that the HackRF has no downsides against it, when properly implemented.
Basically a optimized Power Monitor and the ability to activate a warning sound when reaching a setable threshold.

schmiernippel, 2024-11-11:
In this video you see how the app work in the first picture, maybe it help you with info!

https://youtu.be/mKjc3Z_6IXY

JAKAMI99, 2025-04-12:
👀

htotoo, 2025-05-24:
Started an experiment on this.

htotoo, 2025-05-24:
Memo for myself:

// TETRA Uplink Frequencies to Monitor (in Hz)
// Focused on the 380-390 MHz band (commonly public safety)
// Each frequency represents the center of a 300 kHz monitoring window.

const std::vector<uint32_t> tetra_uplink_monitoring_frequencies_hz = {
    // --- Monitoring centers for 380-390 MHz band ---
    // Each frequency F below means your radio tunes to F, covering F-150kHz to F+150kHz.
    // Band starts at 380,000,000 Hz, ends at 390,000,000 Hz.
    // First center: 380,000,000 + 150,000 = 380,150,000 Hz
    // Last center:  380,150,000 + 33 * 300,000 = 390,050,000 Hz (34 frequencies total for this band)
    380150000, 380450000, 380750000, 381050000, 381350000, 381650000, 381950000,
    382250000, 382550000, 382850000, 383150000, 383450000, 383750000, 384050000,
    384350000, 384650000, 384950000, 385250000, 385550000, 385850000, 386150000,
    386450000, 386750000, 387050000, 387350000, 387650000, 387950000, 388250000,
    388550000, 388850000, 389150000, 389450000, 389750000, 390050000
};

// --- Array for Remote Control Frequencies (433.92 MHz and 315 MHz regions) ---
// Each frequency is the center of a 300 kHz monitoring window.
const std::vector<uint32_t> remotes_monitoring_frequencies_hz = {
    // Around 315 MHz (common for older remotes, key fobs in some regions)
    314700000,    315000000,    315300000,
    // Around 433.92 MHz (very common for remotes, sensors, key fobs globally)
    433920000,
    434010000,
};

const std::vector<uint32_t> lora_monitoring_frequencies_hz = {
    // EU433 Band (Europe, typically 433.05 MHz to 434.79 MHz)
    // Scanning the approximate range 433.0 MHz to 434.8 MHz
    433150000, // Covers 433.00 - 433.30 MHz
    433450000, // Covers 433.30 - 433.60 MHz
    433750000, // Covers 433.60 - 433.90 MHz
    434050000, // Covers 433.90 - 434.20 MHz (includes 433.92 MHz)
    434350000, // Covers 434.20 - 434.50 MHz
    434650000, // Covers 434.50 - 434.80 MHz

    // EU868 Band (Europe, typically 863 MHz to 870 MHz, specific channels around 868 MHz)
    // Targeting common LoRaWAN channel groups
    867200000, // Covers 867.05 - 867.35 MHz (for channels like 867.1, 867.3)
    867500000, // Covers 867.35 - 867.65 MHz (for channels like 867.5)
    867800000, // Covers 867.65 - 867.95 MHz (for channels like 867.7, 867.9)
    868100000, // Covers 867.95 - 868.25 MHz (for channels like 868.1)
    868400000, // Covers 868.25 - 868.55 MHz (for channels like 868.3, 868.5)

    // US915 Band (North America, typically 902 MHz to 928 MHz, specific channels around 915 MHz)
    // Providing a few sample windows around the 915 MHz area.
    // This band is wide; a full scan would require many more frequencies.
    914700000, // Covers 914.55 - 914.85 MHz
    915000000, // Covers 914.85 - 915.15 MHz (Centered on 915 MHz)
    915300000, // Covers 915.15 - 915.45 MHz
};

htotoo, 2025-05-27:
Just built a test version. Will test it, and if it seems stable, I'll publish it

schmiernippel, 2025-05-27:
Great, cant wait to test it......👌

brisioksss, 2025-05-27:
> Just built a test version. Will test it, and if it seems stable, I'll publish it

Lets gooooo

brisioksss, 2025-05-27:
Sweeeeeeeeeeeeeeeeet! Good job man!

htotoo, 2025-05-30:
n_250529 has it under the RX menu with the name Detector

Radiomix2000, 2025-05-31:
> Maybe it'll need some amp/lna/vga / bip tweaking

yep, the bandpass filter is mandatory, otherwise the Rx chain is overloaded from near 4G/5G basestations. HackRF doesn't have it internally.

htotoo, 2025-05-31:
The sw is limited to 750k bandwidth (and jumping), and the ranges seems ok based on that.
But yeah, the receiver part can do strange stuff.

closed 2025-06-05 by htotoo