portapack-mayhem/mayhem-firmware #2137: [Wiki Issue] Wiki is using grabify link to track users

Programicus, 2024-05-04:
### Describe the bug.

Currently, the [H2 mayhem edition on the portapack versions page](https://github.com/portapack-mayhem/mayhem-firmware/wiki/PortaPack-Versions#h2m-mayhem-edition) uses a grabify link to redirect to alibaba. This was introduced in commit ddb6ab2e16c331ca744fab018087849c671be958 to the wiki.

For context, grabify is a url shortener (similar to bit.ly), however it also comes with a page that logs all uses of the redirection. [Here](https://grabify.link/track/MOT632) is an example such a tracking and logs page for a redirect I made to github.com, as well as a hit it via [archive.ph](https://archive.ph/q7PVa) to give an example hit without doxing myself. As you can see from this example, I get quite a bit of information about people who the link: ip address, timezone, browser info, isp, os version, etc. 

### Reproduction

Go to: [H2 mayhem edition on the portapack versions page](https://github.com/portapack-mayhem/mayhem-firmware/wiki/PortaPack-Versions#h2m-mayhem-edition).  Hover over the link and see that it `https://grabify.link/5CL1IP`.

### Expected behavior

Have either a normal alibaba link, or one from a reputable url shortner like bit.ly or t.ly

### Anything else?

Sorry if this isn't the best way to post this. I couldn't find a way to create an issue about the wiki page

eried, 2024-05-05:
Hi! Grabify is made by one of our contributors, I actually prefer this redirection than other option knowing we can discuss directly about it.

What is your concern? this redirection does not get more info that any script can get from your browser, there is no "magic" spyware or something evil behind. We ofc get click stats about what users are interested on.

eried, 2024-05-06:
I added a note on the [wiki](https://github.com/portapack-mayhem/mayhem-firmware/wiki/Redirections). It needs more work, but it is open for edits

closed 2024-05-06 by eried

Programicus, 2024-05-07:
Fair enough that you are just using it for click stats, and makes sense knowing that grabify was made by one of the contributers.

I'm used to it seeing grabify used to with respect to scamming and scam baiting, so and had associate it with a lack of trust in the clicker. You're right in that it doesn't give you anything extra than if you were hosting your the link shortener on a server you controlled.

vorpalhex, 2024-08-09:
This grabify link is broken under Pihole and most forms of adblock since grabify is used for scams and phishing so often. I can't even follow the links because of this.

eried, 2024-08-15:
> This grabify link is broken under Pihole and most forms of adblock since grabify is used for scams and phishing so often. I can't even follow the links because of this.

As we explained, grabify can be consider an "in-house" product, so for the moment there is no plans of removing it. We know there is a small amount of people that get bothered by it, im sorry about that. 

Option: whitelist it in pihole and report it as a wrong url to filter, since it is just coincidental that was used for phishing 😃

vorpalhex, 2024-08-15:
If your goal is to have as many people use your affiliate link as possible, then just expand the affiliate link. Right now most people are going to click it, discover it's "broken" and then leave.

Whereas if you just have the whole aliexpress link with the aff_id in it.. it'll actually still work under pihole/adguard/etc. Afaik the only situation where the link would get stripped is a right click "copy clean link" under Brave.

eried, 2024-08-16:
Good but as I explained, for the moment is no plans of removing the redirection

BeanBagKing, 2024-08-27:
The in house comments don't really play into this for me. It's not in house for me, it's in house for you, and no offense, but I don't know you. Not to reiterate a point too many times, but these kind of trackers are usually associated with scammy behavior. I don't like bit.ly or any of the other shorten either, but at least those can be used with URL expanders.

More importantly, it takes something pretty significant for me to go whitelist something in my pihole. It doesn't interfere with the majority of sites, and when it does I'm typically more motivated to move on than unblock it. It's been added to one of the lists for a reason, so why would I go and exclude it?

eried, 2024-08-28:
Adding it to the whitelist was only suggested if you want to click our links. Until we decide to stop using grabify, it's the only way to decode its secrets 😁

JAKAMI99, 2024-11-07:
I only saw bad actors using grabify in the past, it is really not reputable to use an link shortener, that needs ToS to be accepeted before continuing. This really hurts the rep of this project imho.
Why would you want to log the IPs of the community?! Open source should be all against surveilance and tracking... 

While this still in place I recommend addons like fastforward for [Firefox](https://addons.mozilla.org/en-US/firefox/addon/fastforwardteam/) and [Chrome](https://chromewebstore.google.com/detail/fastforward/icallnadddjmdinamnolclfjanhfoafe?)

htotoo, 2024-11-07:
I saw a bad guy using google drive, to share malware. Also i saw bad guys using github. Even saw youtube "tutorials" spreading malware (download xy to do this).
I think it doesn't really matter what you use, that matter, how you use.

Mayhem don't use it in any abusive way, so it still good.