portapack-mayhem/mayhem-firmware #2149: Readable LOGs

voxo22, 2024-05-14:
### Description of the feature you're suggesting.

Guys, is it possible to fulfill the LOGs (if checked to create) with usefull content? E.g. I have set up logging while radiosonde passed my view, but I realized I cannot do anything with the logged data. Just bunch of hexcode... Or how can I decode all the info from it?
Thank you

voxo22, 2024-06-01:
Ok, I did the research from the original code for both RS41 and M20 and made simple PHP website able to decode the radiosonde telegram from the log. For me now it's acceptable solution.

htotoo, 2024-06-02:
If you could extract any new data from the hex logs (that is not displayed in pp), could you share your code please?

voxo22, 2024-06-03:
[M20-RS41.zip](https://github.com/user-attachments/files/15529593/M20-RS41.zip)
Ok, here it is, just decompress and run in Wamp, or webserver. File SONDE.TXT should be in the same folder as .php file

closed 2025-09-09 by htotoo