portapack-mayhem/mayhem-firmware #2150: ADS-B has no geo for airplane

danaaaadash, 2024-05-14:
### Describe the bug.

Hi, just understood that something goes wrong...
In RX ADS-B app i can see packets, but when i choose any of them, i dont any inf about speed, lvl, alt, lan and long coordinates. when i choose "see on map" i see just point at lat: 0. 0 0 0... long: 0. 000000 alt: ---- and etc. I tried defferent antennas and different firware version but i still have this problem. Please help me someone, i didn't find any inf about this in internet....

portapack h2

### Reproduction

1. sw on
2. recive menu
3. ADS-B app
4. any ADS-B packet

### Expected behavior

see the gps point of airplane

Yzord, 2024-05-15:
Did you placed a sd card in it with the right files on it?

danaaaadash, 2024-05-15:
> Did you placed a sd card in it with the right files on it?

yes, of course

Yzord, 2024-05-15:
And you are sure that the sd card can be read by the portapack? And how did you placed the files on it? Because something is wrong there.

danaaaadash, 2024-05-15:
> And you are sure that the sd card can be read by the portapack? And how did you placed the files on it? Because something is wrong there.

YES I did it all, I checked all files for ads-b and I tried to format sd and rewrite files (FAT32) but it didnt help. i tried to take other sd,  but still having same problem. 
I tried:
different firmwares
different antenas
different sd
different locations for reciving (in different locations in city, near buildings, far from buildings, outside the city, in forest on big clear field and lot more)

NotherNgineer, 2024-05-15:
What are your LNA, VGA, and Amp settings?  I usually use 32, 32, and 0 respectively.  Maybe you just need the airplanes to fly closer or turn up the gain settings.

Even without an SD card installed, the ADSB RX app should show level and speed on the main page, and altitude, latitude, longitude, heading, speed, and "sil" on the Details page.

Usually it takes a few packets from the same plane before the data shows up.  With a weak signal you might never see these details.

gullradriel, 2024-05-15:
Is your CPLD up to date ?
Is external TXCO NOT disabled in settings/radio ?

gullradriel, 2024-05-31:
Closing due to inactivity/enough answers provided.

closed 2024-05-31 by gullradriel